particle-iot/device-os #821: System threading on Electron sometimes crashes

m-mcgowan, 2016-01-18:
System threading on the electron fails just after the cloud connection has been established. 

I've investigated briefly, changing thread stack sizes to rule out stack overflow. The error happens in the FreeRTOS queue code, so looks like one of the thread queues is being corrupted.

I'll follow up with steps to reproduce the issue.

m-mcgowan, 2016-01-19:
I've yet to verify, but I'm also seeing problems on the Photon due to newlib-nano not supporting thread-safe heap allocation. #826

mhazley, 2016-01-20:
@m-mcgowan is this in anyway related to the issue I raised [here](https://community.particle.io/t/v0-4-8-rc1-wifi-reconnection-issue/18904)? Just curious as was going to raise my thread trace on github but will leave it if this in a similar court.

m-mcgowan, 2016-01-31:
I don't believe the issues are related - is there a connection you see between them?

mhazley, 2016-01-31:
Not overly, my only connection was that my system thread seems to die (can't break on `system_thread_idle`) after a number of reconnects - sounded like this might have been related.

PR #884 (2016-03-10): Increased stack size and stack overflow detection for Electron
This PR fixes #821 by increasing stack size for threads created via `std::thread` wrapper and adds FreeRTOS hook for stack overflow detection -- latter is enabled only for debug builds on Electron currently (thanks to @avtolstoy for pointing me out).
closed 2016-03-10 by m-mcgowan