particle-iot/device-os #823: User program slows, stalls or crashes when a delay is present in Listen mode

karlhenricksen, 2016-01-18:
### Description: 
Using the latest develpment version 0.4.8-rc.6, we narrowed down some issues around Listening mode to a simple delay (see sample pgm) - when entering listening mode, our real user program was crashing due to the delay we had used to try to debug a separate issue. However, this delay should not be problematic. Currently it causes a slow blue blink or steady cyan LED and the user pgm to slow/stall or crash when entering or sitting in listening mode.

### Sample Program to Reproduce:
```cpp
#include "application.h"

SYSTEM_THREAD(ENABLED);
SYSTEM_MODE(SEMI_AUTOMATIC);

void setup()
{
  Serial.begin(9600);

  WiFi.on();
  WiFi.connect();
  Particle.connect();

  Serial.println("Attempting to cause listening mode crash");

  WiFi.listen();
}  

void loop()
{
  delay(2000);
  Serial.println("Still here...");
}
```

### Steps to Reproduce:
1. Load the attached application to Photon
2. With Photon in listening mode, note the serial debug output - should be printing every 2s, but starts slowing after a few seconds and LED slowly blinks blue --> see video 1 below
3. Set up Photon wifi with valid creds to connect to cloud (breathing cyan)
4. Hold Setup button for ~4s to enter listening mode - user app crashes and serial debug stops, LED holds steady cyan --> see video 2 below

### Videos of Bad Behavior:
Video 1: Slow blink blue LED / slow or stalled user prgm 
https://www.youtube.com/watch?v=szIjvjSxSgQ

Video 2: Steady cyan LED / user crash when entering Listening Mode:
https://www.youtube.com/watch?v=Nck-RoLtJBI

m-mcgowan, 2016-01-18:
Thank you for the test case. We have been improving the interaction of threading and listening mode, so going to try this against the current develop state now.

m-mcgowan, 2016-01-18:
Please try against commit 6624fdbb424a52df003d394d944c5e4bba7fde5c which contains fixes to both how application and system threads operate during listening mode. I was unable to reproduce the crash on that commit.

karlhenricksen, 2016-01-19:
@m-mcgowan I tried against the latest commit, and the first behavior (step 2 / video 1) is no longer seen, but it's still easy to crash the app by going online and entering listening mode (step 4 / video 2). Just follow the steps to the end and you should easily reproduce. 

Perhaps related, I'm also seeing the 2sec delay is not actually 2s with the latest, it prints anywhere from 1 to 3sec.

m-mcgowan, 2016-01-22:
Thank you for the clear test case - it helped in quickly identifying and resolving the issue.

commit bc7d59b6 (2016-01-22): fix #823.  Adding to a queue may happen from an ISR when it's the result of a button event. If the queue is already full, then calling the non-ISR method will block indefinitely since the scheduler cannot run to wake up the task.
closed 2016-01-22 by m-mcgowan