particle-iot/device-os #825: SoftAP unresponsive after entering listening mode for the second time

indraastra, 2016-01-18:
This is probably related to issue #788 but I'm starting a separate thread since the bug doesn't result in the user or system code being blocked.

I'm running [this code](https://gist.github.com/indraastra/b25f5b2072cbc3b572f4) with the 0.4.8-rc.6 firmware and I consistently run into an issue with the following series of steps:

(this is all in `SEMI_AUTOMATIC` mode with the firmware linked above)
1. Delete wifi credentials by holding mode button for 10 seconds
2. Restart Photon, device enters listening mode due to setup calling `Particle.connect()`
3. Configure credentials => connection should succeed!
4. Hold SETUP button to enter listening mode again (this time explicitly)
5. Configure credentials => setup fails because device is unresponsive to HTTP calls; console shows `GET http://192.168.0.1/device-id net::ERR_CONNECTION_REFUSED`, and indeed, trying to view this page fails in Chrome with the same error.

So glad you're digging into these issues! We'd love for this part of the setup to be as smooth as possible.

indraastra, 2016-01-18:
Ok, something's changed ever so slightly as of commit 6624fdb. To recreate, amend step 4 from above to:

4. Hold SETUP button long enough to clear credentials and enter listening mode again

You can pretty much infinitely reproduce this by resetting after step 5: listen, configure, delete, attempt to re-configure, fail.

m-mcgowan, 2016-01-18:
I couldn't reproduce this against the current develop branch . Daft question, but just to be sure - can you confirm you are reconnecting to the Photon's SoftAP after step 4?

indraastra, 2016-01-18:
Hmm, very odd. I've tried with both a desktop and a phone and while I'm assigned IPs in the 192.168.0.x subnet, I'm not able to fetch any of the web pages:
```
$ ifconfig
en1: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	inet 192.168.0.2 netmask 0xffffff00 broadcast 192.168.0.255

$ curl http://192.168.0.1/device-id
curl: (7) Failed to connect to 192.168.0.1 port 80: Connection refused
```

indraastra, 2016-01-18:
Just to be 100% clear, the sequence I'm running through is:

1. Start device with no credentials, should enter listening mode
2. Configure, should connect fine
3. Clear credentials with setup button, should enter listening mode
4. Attempt to reconfigure, should fail

The serial output does reflect every state change except the loss of credentials from step 3, which isn't really that important, just interesting.

m-mcgowan, 2016-01-18:
Thanks. Which apps on your desktop and phone are you using to configure credentials via HTTP? I should try replicating your setup as close as possible.

indraastra, 2016-01-18:
Ack, looks like the difference in our test cases was probably introduced by me when I uncommented this line:

`UPDATE_IF_CHANGED(state.wifi_has_credentials, WiFi.hasCredentials(), "* Wifi hasCredentials")`

I added this because it could be useful in inferring what step of the setup process the user is on, and removed it due to some race condition I never fully understood.

Anyway, when I comment it out again I'm able to clear and reconfigure the device a few times in a row  without resetting. (For reference, I've been using the excellent http://photonsoftap.meteor.com/ )

indraastra, 2016-01-18:
I guess it's still an issue, though perhaps an expected one. What I'd love to accomplish is to completely insulate our product's users from the Photon's inner workings by lifting out the connection state from the system to the level of our application. Knowing if there are credentials isn't as important to us now that we can safely call `Particle.connect()` and have it do the right thing without blocking, but it may still be useful in tracking a user's progress through the setup process and to give them relevant feedback.

m-mcgowan, 2016-01-18:
While I can't replicate the issue exactly as you describe it, I do have 2 related issues:

- repeatedly bringing up SoftAP shows a memory leak of around 2Kbytes per invocation. 
- including `UPDATE_IF_CHANGED(state.wifi_has_credentials, WiFi.hasCredentials(), "* Wifi hasCredentials")` in the code causes a hard fault. 

I'll look into these as soon as I can. 

In the meantime, a workaround for the memory leak is to listen for the `wifi_listen_end` event and reset the device.  Not an ideal solution, but may bring more stability until we have found the source of the leak.

m-mcgowan, 2016-01-22:
Could you try against the latest development state please https://github.com/spark/firmware/commit/bc7d59b65bb8e22714b8d0186c5023747b91a10f I fixed the memory leak and the SOS that I was observing, which I hope improves the behavior you were seeing.

indraastra, 2016-01-26:
Sorry for the delay on this, but I just tested a minute ago and I'm still seem some kind of variant of this issue. I'll do some more testing to see if I can narrow it down a bit more first.

indraastra, 2016-01-26:
From what I can tell, the credential setup process always succeeds when initiated from the particle cli even after multiple rounds of clear & configure during a single startup. However, as soon as the the photonsoftap.meteor.com client is used to configure credentials once, something seems to get knotted up in the system firmware. My naive guess is that it has something to do with the multi-packet HTTP request fix, but I'm not 100% sure.

indraastra, 2016-01-26:
Here's another way to trigger this:

1. Clear credentials
2. Configure credentials using photonsoftap.meteor.com -> should work
3. Repeat 1-2 -> should work
4. Repeat 1-2 -> browser console should show failed http request for /device-info
5. Configure credentials using `particle setup web` -> succeeds
6. Clear credentials -> LED never blinks blue like it would upon entering listening mode but AP is definitely up and available

m-mcgowan, 2016-03-18:
I tried the steps today - I was able to repeat steps 1+2 5 times without error. I was using system firmware from commit 0384e4609035fcf7b954607b0f97f246ec24cbf4.