particle-iot/device-os #828: Enhancement: Leave the manage_network_connection() to do the network connection, instead of before listening mode exited

XuGuohui, 2016-01-20:
Hi guys,

When I enabling the serial debug, I just noticed that before smart config exiting, it tries connecting to the AP once in template function `start_listening()`: 

`
if (has_credentials())
            connect();
        else if (!started)
            off();`

 If failed because of wrong password, it will try connecting to the AP again, launched by `manage_network_connection()` and definitely it must fail to enter the listening mode. 

How about just exiting the listening mode after WiFi credentials being stored and let the `manage_network_connection()` to do the rest work?

Thank you

m-mcgowan, 2016-01-29:
If I'm understanding your point, you'd like to not connect after credentials are entered?

I can understand the need for this - I've seen the system behave unexpectedly when in MANUAL mode. When it enters listening mode and credentials given, the system then automatically connects to the network. This goes against the intent of manual mode, where the system leaves all connection initiation to the application.

However, the converse needs to also be true - if the network was connected on entering listening mode, then it makes sense that it remains connected when exiting listening mode. even in MANUAL mode. The `manage_network_connect()` call won't automatically activate the network in MANUAL mode, so it must be the responsibility of the listening mode to restore the state. 

With this in mind, it would make sense that existing listening mode should attempt to restore the state the network as it was when listening mode was started. The three possible states are: Off, On, and Connected.

How does that sound?

m-mcgowan, 2017-02-07:
on hold pending clarification and feedback

Cheong2K, 2017-09-26:
hi @m-mcgowan, yes, I can repeat this problem. In user part, if I set to MANUAL mode and then I go into the listening mode, after entering credentials, it exits and connects to the AP I set, but my user part is just a simple blinking without any connectivity code.

I will take a look and try to fix this behaviour.