particle-iot/device-os #831: analogWrite() : Taking care of the passing in parameter 'value' in the wiring function

XuGuohui, 2016-01-22:
Hi there,

For users using the analogWrite() API, they probably pay no attentions on the passing in parameter `value`, which range depends on the parameter `pin`. If using the pins equipped with DAC, the `value` ranges from 0 to 4095, while using the other pins equipped with timer, the `value` ranges from 0 to 255.

How about unifying the range of the passing in `value` in the HAL layer so that users just only remember  one range of the `value`, no matter which pin they want to use?

Thanks!

m-mcgowan, 2016-01-29:
I could imagine using a `float` value from 0 to 1.0 that represents minimum and maximum output. This essentially provides resolution independent setting of an output value, whether it's digital, PWM or DAC. 

It's a nice idea having the pins consistent, but I wonder if it would cause confusion?

An alternative approach would be to log an error if the pin values are out of range.

XuGuohui, 2016-01-29:
The first method you bringing up sounds great. 

I don't know if someone who hasn't viewed the source code knows the value range on different pin, especially using DAC. My colleague just made this mistake when output analog value on DAC pins - He passed in 255, the value he thought being the maximum value, while the LED worked nearly weakly.

m-mcgowan, 2016-01-29:
You don't need to look at the source - the pin parameters are described in the docs - https://docs.particle.io/reference/firmware/photon/#analogwrite-pwm-

XuGuohui, 2016-01-29:
Awesome! It describes very in detail. 

Thank you!

closed 2016-01-29 by XuGuohui