particle-iot/device-os #832: Extra comma in interrupts_irq.h

ccFarias, 2016-01-22:
There is an extra comma at line 133. It makes SysInterrupt_CAN1_SCE_IRQ undefined and sets the wrong value to __Last_irq.

m-mcgowan, 2016-01-23:
Thanks for taking the time to report the issue. I'm not sure I follow you - I added a static assert to validate that both __Last_irq is defined and equal to  SysInterrupt_CAN1_SCE_IRQ+1, and verified the value is 83.

bkobkobko, 2016-01-23:
Hi @m-mcgowan 

I think the OP is talking about the comma after the `+ 1,` in the last line of the enum.  If you follow the macro expansion, I think you will have `the_last_element, }` which is legal syntax in C99 and up (but not before).

m-mcgowan, 2016-01-25:
Thanks for that bko,  Given that we are targeting C99/C++11,  I'm not seeing the issue. 

Of course, it's no problem to remove the comma, but it would be good to know what problems it's causing for the OP.  @ccFarias can you provide more details on what the problem this is causing for you?

ccFarias, 2016-01-26:
I didn't know that was valid C/C++. In C# that does not even compile. Anyway, something else is wrong with this enum. I am using Particle Dev and if I print *__Last_irq* (`Serial.println(__Last_irq);`) using the serial monitor, it prints out 46. Besides, if I try to call *attachSystemInterrupt* using values from *SysInterrupt_CAN1_TX_IRQ* to *SysInterrupt_CAN1_SCE_IRQ* it gives me a compilation error ('XXX' was not declared in this scope) and no interrupts with values greater than 45 passes *is_valid_irq* validation. This was already mentioned [here](https://community.particle.io/t/strange-behavior-with-spi1-irqhandler-solved-bug-report/15180/8).

m-mcgowan, 2016-01-26:
Sounds like you're targeting the Core in your build.

ccFarias, 2016-01-26:
It sounds like I am targeting the Core, but I am not. I only have the Photon and it shows as the target device at the bottom of the screen. If I had the Core, I wouldn't be able to compile this `attachSystemInterrupt(SysInterrupt_TIM5_Update, newHandler);` right? *SysInterrupt_TIM5_Update* It is defined only for the STM32F2XX.

monkbroc, 2016-02-10:
I fixed this issue in the CAN driver commit:
https://github.com/spark/firmware/commit/9b6f66b6ec6c1382cab8a5cf818475cea8c7a507#diff-27b4f75c5ca24066a9ee22fcb4d460b8L129

__Last_Irq was hardcoded to 46.

closed 2016-03-16 by m-mcgowan