particle-iot/device-os #833: DAC1/2 possible problem with digitalWrite() after analogWrite() [0.4.9]

ScruffR, 2016-01-23:
As it seems after `analogWrite()` on a DAC pin `digitalWrite()` does not work anymore.

```cpp
SYSTEM_MODE(SEMI_AUTOMATIC)

void setup()
{
    pinMode(D7, OUTPUT);
    pinMode(A6, OUTPUT); // DAC1
    pinMode(A3, OUTPUT); // DAC2
}

int i;
void loop()
{
  if (0 <= i && i <= 4095)
  {
    analogWrite(A6, i);
    analogWrite(A3, 4095-i);
  }
  else
  {
    digitalWrite(A6, i>0 ? HIGH : LOW );
    digitalWrite(A3, i>0 ? HIGH : LOW); 
  }
  i++;
  if (i>=5000)
  {
    i=-1000;
    digitalWrite(D7, !digitalRead(D7));
  }
  delayMicroseconds(20);
}
```
More tests and scope will follow

m-mcgowan, 2016-01-23:
Internally the pinMode is changed to enable the DAC. Could you add a pinMode() call before digitalWrite() to see if that fixes it?

If that fixes it, I can add a check in the HAL for the current pin mode as part of digitalWrite, and if it's the DAC pinmode, change it to OUTPUT.

ScruffR, 2016-01-23:
I had tried that but it didn't change anything. But next I'll try a `pinMode(DAC, INPUT); pinMode(DAC, OUTPUT);` combo to see if this changes anything.

PR #855 (2016-03-10): Added HAL_DAC_Enable()/HAL_DAC_Is_Enabled methods to control DAC state on pin
Fixes #833
commit 3e7e8885 (2016-03-10): Added HAL_DAC_Enable()/HAL_DAC_Is_Enabled method to control DAC state on pin. Fixes #833
closed 2016-03-10 by m-mcgowan