particle-iot/device-os #840: spark_wiring_spi.h:115:32, "expected unqualified-id before ')' token"

Jerware, 2016-01-26:
The 0.4.9 firmware has an issue compiling my FastLED sketch. I get six repeated errors that say, "expected unqualified-id before ')' token in ../wiring/inc/spark_wiring_spi.h:115:32

I also get 30+ compile errors, all pointing to chipsets.h inside FastLED complaining that "mSPI was not declared in this scope." I'm hopeful these errors are due to the system SPI problem.

m-mcgowan, 2016-01-26:
SPI was changed from a global instance to a #define to ensure that it is initialized before use. 

I'm guessing the library is defining it's own SPI symbol?
If you are not using SPI, then you can add #undef SPI before including other application headers.

Jerware, 2016-01-26:
Thanks, you're absolutely right. I've raised the issue on the FastLED GitHub.

m-mcgowan, 2016-01-28:
Reverted the change of SPI to a macro.

closed 2016-01-28 by m-mcgowan