particle-iot/device-os #841: Allow changing the PWM and DAC resolution

monkbroc, 2016-01-27:
Add analogWriteResolution to allow changing the PWM and DAC resolution from 8 bits to up to 16 bits (that many bits may not be available on the hardware).

Community post http://community.particle.io/t/photon-pwm-resolution/19395/9

See https://www.arduino.cc/en/Reference/AnalogWriteResolution for specification

m-mcgowan, 2016-01-27:
Relates to #213, which closed when `analogWrite()` was extended to allow the frequency to be set.

monkbroc, 2016-01-27:
I missed that #213 was asking for both resolution and frequency improvements :flushed:

andyggraves, 2016-03-19:
Hi, everyone, not meaning to push, but was wondering what is the process / how long for an idea like this? Cheers

BasedOnTechnology, 2016-03-22:
@m-mcgowan Echoing @andyggraves. This would be really, really helpful for me too.

monkbroc, 2016-03-23:
I hear you. This is something I have on my list of things to do.

BasedOnTechnology, 2016-03-23:
@monkbroc Thank you very much!

andyggraves, 2016-05-27:
This is wonderful, thank you.
Will it become available on the web build UI?

andyggraves, 2016-05-30:
Hi Guys, 
I am trying to use the new exciting analogueWriteResolution function but it does not seem to be recognised "particle build" where am i going wrong?
Thanks
Andy

ScruffR, 2016-05-31:
This hasn't been merged into the release branch yet -  there are still some tasks to be completed first.

PR #991 (2016-06-10): DAC and PWM resolution
Closes #841 

This PR implements configurable resolution for analogWrite (PWM and DAC).

New Wiring functions:
- `uint8_t analogWriteResolution(pin_t pin, uint8_t value);`
- `uint8_t analogWriteResolution(pin_t pin);`
- `uint32_t analogWriteMaxFrequency(pin_t pin);`

(see https://github.com/spark/firmware/blob/feature/dac-pwm-resolution/docs/reference/firmware.md for documentation)

DAC on Photon and Electron supports only 12-bit (default) or 8-bit resolution

Resolution also affects maximum frequency that can be used with `analogWrite()` on PWM pins:
## Supported resolu [... cut]
closed 2016-06-10 by technobly