particle-iot/device-os #842: Particle.variable(float) produces odd result (float gets treated as double)

ScruffR, 2016-01-27:
https://community.particle.io/t/string-tofloat-oddities/19530/2

Despite the title of the topic the problem is not `String::toFloat()` but as it seems that you can pass a `float` to `Particle.variable()` but the cloud side of things tries to take it as `double` 

```cpp
float foo;
Particle.variable("Foo", foo);
```
is reported by `particle list` as 
```text
  Variables:
    Foo (double)
```

closed 2016-01-28 by m-mcgowan

ScruffR, 2016-01-29:
Is this meant to be the final word on this or just a workaround for the time being?

What would `Particle.variable("foo", (double)floatVar)` do?

The "new" two-param overload somehow obscures what datatypes are supported, and I thought this was meant to support arbitrary datatypes after all.

m-mcgowan, 2016-01-29:
Github closes issues sometimes based on commit comments. It's not intended to be a final word - is there ever a final word with continually evolving systems?

The commit is there to directly catch cases where float is used.  Sure, casting to a double passes the compiler checks for now, although I'm surprised it doesn't complain about taking the address of a temporary. 

We will be revising the inner workings of variables soon. I'll see if we can have float support added to the cloud since this topic has come up several times before and seems to be a source of confusion.

ScruffR, 2016-01-29:
Yup, no build message with the cast but still the same result.

m-mcgowan, 2016-01-29:
We aill add 2 changes to variables:

- on demand computation of the variable value, for things like WiFi.localIP()
- detection of when a variable is on the stack -  the registration will not succeed, to prevent a potentially difficult situation.

commit c192999b (2016-01-28): produces a compiler error when `Paritcle.variable()` is called with a `float`. Fixes #842
commit 035cf2f5 (2016-02-01): produces a compiler error when `Paritcle.variable()` is called with a `float`. Fixes #842