particle-iot/device-os #844: [Photon] factory reset does nothing

kennethlimcp, 2016-01-28:
Using the latest bootloader version 7, performing a `System.factoryReset()` simply resets the Photon without performing the following:

```
-- Restore factory reset firmware from external flash (tinker)     <----this should be removed from the docs for Photon and electorn?
-- Erase WiFi profiles
-- Enter Listening mode upon completion
```

technobly, 2016-01-28:
@kennethlimcp did you try flashing tinker to the factory image location first? `particle flash --factory tinker`

kennethlimcp, 2016-01-28:
@technobly i remember reading something about factory image area no longer available right? Has that changed now?

technobly, 2016-01-28:
By default there is no factory image loaded at 0x080E0000 on the Photon/P1, but if you put one there with `particle flash --factory tinker|firmware.bin` you can use buttons or System.factoryReset() to restore.  I just tried with the buttons on my Photon and it worked and reset my WiFi profiles.  Looks like the docs need updating again :)  Thanks for pointing this out!

kennethlimcp, 2016-01-28:
I'll try it in the evening :)

technobly, 2016-01-28:
Note: This is only for user-part modules though, as there is only 128k available for factory images.

m-mcgowan, 2016-01-28:
Manufactured Photons do have the factory reset image, and so factory reset will work as expected.

This image is overwritten when performing a system update (but not an application update.)

I've been thinking about how to restore the factory image after a system update. There are some options:
- an OTA (initiated by the CLI, Dashboard) that is flagged as "factory default" so the image is stored in the factory default location
- a special event which when sent to the device makes the current application also the factory default by copying to the factory backup location
- a "factory" image on the application so that this is automatically made the factory default (and used to restore the factory reset area after a system upgrade.)

Just thinking aloud here - thoughts and use cases welcome.

kennethlimcp, 2016-01-28:
I would vote for --> an OTA (initiated by the CLI, Dashboard) that is flagged as "factory default" so the image is stored in the factory default location

Since this makes more sense for the product creator...

m-mcgowan, 2016-04-14:
The destination storage medium can be given as part of the OTA protocol. Presently only `FIRMWARE` is supported, meaning store to main firmware. https://github.com/spark/firmware/blob/develop/communication/src/chunked_transfer.cpp#L38

https://github.com/spark/firmware/blob/develop/system/src/system_update.cpp#L208

A new storage `FACTORY_FIRMWARE` will be created.  `Spark_Finish_Firmware_Update` calls `HAL_FLASH_End` - this will need to be instructed to store the downloaded image in the factory reset location.

m-mcgowan, 2017-07-07:
Just reviewing this again, and I'm curious what the use case is that requires the factory image to be OTA updated, given that a factory reset typically requires physical presence.

kennethlimcp, 2017-07-07:
Probably outdated with all the system/bootloader OTA updates

closed 2018-12-24 by kennethlimcp