particle-iot/device-os #846: Photon System.sleep(pin, edge, seconds) does not always stay asleep

technobly, 2016-01-28:
It's very bizarre... at first I thought I was seeing things.. but it appears to come and go.  I was using INPUT_PULLDOWN on D0, with RISING edge... and even with an additional 1k pulldown resistor it still misbehaves.  Here's a test app that will connect, wait 5 seconds, publish a message, then sleep for 30 seconds and repeat.  It does not always stay asleep for 30 seconds.  I also tried jumpering VBAT to 3V3 which is important for some `retained` variable cases (not that this should affect how long the STOP mode stays asleep).

Tested on latest develop:
https://github.com/spark/firmware/commit/04ededb8ce4d0911f75467adfb506ae0dc419441

Test app:
https://gist.github.com/technobly/b64574b0f31e598c1560

If the app doesn't seem to be failing, try doubling the delays and use a clean build.

technobly, 2016-03-04:
- [x] I tested this recently with #864 and merged #845 on an Electron and sleep stop mode was working.
- [ ] Test with a Photon.
- [ ] Also verify the problem case again with above steps.
- [ ] If anyone else can reproduce any or all of the above that would be helpful :)

m-mcgowan, 2016-03-16:
@technobly - could you complete the remaining test steps please so we can close this issue before the of the week.

technobly, 2016-03-17:
@m-mcgowan completed tests, looks good as far as PR #864 and #845 currently sit.  We should test again after beta 0.5.0 is out.

closed 2016-03-31 by technobly