particle-iot/device-os #852: Disconnections and fast blinking cyan problem

dguerizec, 2016-01-29:
Sometimes the Photon gets disconnected and cannot reconnect, with the fast blinking cyan syndrome.

I've been able to reproduce the problem accurately when sending more than one message a second, and I've been able to track the bug down to communication/src/buffer_message_channel.h BufferMessageChannel::create() and get a working fix.

Here is the [PR](https://github.com/spark/firmware/pull/851)

m-mcgowan, 2016-01-29:
Thank you for taking the time to submit a PR!

The PR does address an oversight - that the buffer does use a larger offset than necessary. Well spotted!

I'm fairly certain that this isn't the cause of your problems on the Photon though,  because that code isn't used on the Photon - it's part of the new protocol that's used on the Electron. The Photon uses `SparkProtocol`, which manages the buffer as the `queue` array member in the class. 

You mentioned that this happens when publishing frequently.  In terms of filling up the buffer, the amount of buffer used isn't related to how often messages are sent - each message is sent in isolation so they are not stacked up in the buffer. So there's no more risk of overrun sending 1 message or 100 messages per second.  Because of this, using an offset of 2 or 16 isn't really significant - the buffer is a good deal larger than any messages we send. 

If you like any help navigating the code, let me know. It would be great to fix the issue that you're seeing. If it is a buffer overflow, diving into `SparkProtocol` and the use of `queue` would be a good place to start. Good luck!

dguerizec, 2016-01-29:
You're right, just while I received your comment my photon stopped working again :/

Back to the drawing^Hdebugging board then.

Actually I don't know if it's a buffer overflow, but this prefix/suffix has struck me as the possible cause for this problem. I just didn't check if that code was useful or not for the photon.

Anyway, the problem seems more easy to trigger when I'm sending many messages than when I'm sending only a few, so I will keep tracking it this way.

Thanks for the merge.

m-mcgowan, 2016-01-29:
If you have a test app that I can use to reproduce the problem, I can dive in too, if you'd like a second set of eyes on the problem.

dguerizec, 2016-01-29:
I've put a test case [here](https://github.com/dguerizec/test_blink_cyan)

It could be simpler, but I use it to abuse and fuzz the Photon, hence the different settings available.

Run it as-is, and the photon will behave almost normaly (it still disconnects but reconnects quickly).

Use the commands in the README and you can crash it in less than 30s (tested with develop branch).
It looks like there is a race condition somewhere in the code between the sending and the reconnection.

Note: I'll be offline until Feb 8th, so I could probably not handle any questions until then.

dguerizec, 2016-02-11:
Did you managed to reproduce the bug yet ?
Can you confirm it's a real bug in the firmware or is it a bug in my test app ?

m-mcgowan, 2016-08-09:
Could you try with 0.5.2 - I believe there were bugs fixed there that may also address this issue.

dguerizec, 2016-08-23:
Hello, sorry for the delay, but it's been running for one week now with no problem.
I think I'll have to wait a bit more to be sure, since in the past it used to run many days in a go, then crashed several times in a couple hours. §I'll keep you posted if it happens again.

Anyway, thanks for the fix!

m-mcgowan, 2016-09-13:
Going to close this out as fixed - feel free to let us know if the problem recurs and we'll re-open the issue.

closed 2016-09-13 by m-mcgowan