particle-iot/device-os #853: discordant documentation and behavior for TCPServer

akcom, 2016-01-31:
The TCPServer documentation [here](https://docs.particle.io/reference/firmware/photon/#tcpserver) notes the following for write (emphasis added):
> Write data to **all the clients** connected to a server. This data is sent as a byte or series of bytes.

but TCPServer only retains a reference to the most recently connected client in the call to TCPServer::available() thus only the most recently connected client will receive the message.

m-mcgowan, 2016-01-31:
Thanks for bringing this up. The documentation is incorrect in this instance. 

The documentation is editable if yo'd like to make the change yourself.

akcom, 2016-01-31:
Since I don't imagine there is much functional advantage to writing to the last client, is it reasonable to take this function out or modify the code to maintain a client list so it can function according to the documentation?

m-mcgowan, 2016-02-01:
This is pretty tricky to implement - it would mean that each TCPServer instance would keep track of all the currently active TCPClient instances.  This would cause quite a bit of overhead, both time and memory. 

The Arduino EthernetServer does this by simply iterating over the range of socket values and finding the local port that corresponds to each socket, to find the one that matches the server. A new HAL function could be added to do this in a way that makes sense for each network stack. 

To my mind, the only reason for doing this would be for complete compatibility with EthernetServer, which TCPServer is modelled after. I'd like to balance that against the fact that sending data to all open TCP Clients doesn't seem like a particularly useful API imho. The existing API can also lead to resource leaks - #660. What do the community think? Do we try to optimize compatibility at the expense of efficiency?

closed 2016-04-14 by m-mcgowan