particle-iot/device-os #854: I2C does not insure stop condition completes correctly in endTransmission

greiman, 2016-01-31:
At about line 560 of hal/src/stm32f2xx/i2c_hal.c a stop condition is issued but the code does not wait for the stop condition to complete.

This can cause a failure in the next start/send address operation.  The failure depends on the time between the endTransmission() call and the next requestFrom() call.

If the calls are back-to-back, they complete correctly if no extra delay occurs.  This is why adding  SINGLE_THREADED_SECTION() to Wire fixed problems like this issue: https://github.com/spark/firmware/issues/698.

It appears that with the SINGLE_THREADED_SECTION()  back-to-back calls act like a restart condition.

If there is a long delay, the stop condition completes OK and there is no problem.

If there is some delay, the result of the requestFrom() can be a failure with just a start condition followed by a stop condition.

The above issue occurs when these calls are made in a tight loop in the MCP23017 library.

```
Wire.endTransmission();
Wire.requestFrom(MCP23017_ADDRESS | i2caddr, 1);
```
I removed SINGLE_THREADED_SECTION() from spark_wiring_i2c.cpp and ran a test program. It fails in a matter of seconds.

I connected a digital scope and caught the failure
[image]
The transfers up to the gap in SCL is a successful call to endTransmission()/requestFrom(). It is followed by a successful call to endTransmission() followed by a call to requestFrom() that is only a start condition followed immediately by a stop condition.

After looking at the code in i2c_hal.c, I found that there is a problem at about line 560 where endTransmission() does not wait for the stop condition to complete.

I did a quick hack with this fix from "BEGIN WHG ADD" to "END WHG ADD".
```
/* Send STOP Condition */
if(stop == true)
{
      /* Send STOP condition */
      I2C_GenerateSTOP(i2cMap[i2c]->I2C_Peripheral, ENABLE);
      /* BEGIN WHG ADD  */
      int timeout = 0X1000;
      while (i2cMap[i2c]->I2C_Peripheral->CR1 & I2C_CR1_STOP) {
        if (--timeout <= 0) return 6;
      }
      /*  END WHG ADD */
}
```
I tried the test program again without SINGLE_THREADED_SECTION() and it ran for hours with no failures, more than 40 million times through the loop. This is thousands of times longer than the unmodified code.

This is an ugly bug. If the calls to endTransmission()/requestFrom() are back-to-back with no context switch, the bug acts like a restart and the calls are successful. If there is a large gap between calls, the stop condition completes OK and the calls are successful. If there is a slight gap in time, you get the start/stop failure.

The bug is still there even though SINGLE_THREADED_SECTION() reduces its impact for some applications. I was able to put a slight delay between the calls and make it happen even with SINGLE_THREADED_SECTION().  

Also changing the clock rate has an impact on the failure rate.  I have an example that works at 400 kHz but fails at 100 kHz with  SINGLE_THREADED_SECTION().

I suspect I2C failures are occurring in various applications.

m-mcgowan, 2016-02-01:
Thank you so much for finding this! :+1: I've never been entirely convinced that disabling task switching, was the ideal solution, particularly since the WICED I2C driver operates without task switching. 

I'm unsure if the timeout should be based on the number of iterations rather than a wall-clock time? The rest of the I2C uses the number of elapsed microseconds to timeout waiting for an event - do you see any reason not to do that when waiting for the STOP condition?

greiman, 2016-02-01:
> I'm unsure if the timeout should be based on the number of iterations rather than a wall-clock time? The rest of the I2C uses the number of elapsed microseconds to timeout waiting for an event - do you see any reason not to do that when waiting for the STOP condition?

Wall clock should work.  i2c_hal.c has 100 ms timeouts so it should be safe.  My retry count is only about 500 μs.

I have been doing multi-threaded systems since the 1970s and started using retry counts some places in priority based systems.    Retry counts work in interrupts. 

You can have a low priority thread timeout in a busy system without getting enough CPU.

Of course wall clock timeouts have their place also.

PR #856 (2016-03-14): I2C fixes
Fixes #854
closed 2016-03-14 by m-mcgowan