particle-iot/device-os #859: Entering listening mode much slower using WiFi.listen() rather than button press

pomplesiegel, 2016-02-04:
Running 0.4.9 in `SYSTEM_THREAD(ENABLE)` we're often seeing delays of three minutes or more following the call to `WiFi.listen()` before the Photon actually enters listening mode. In contrast, a "setup" button hold causes the device to immediately begin listening. Is there a reason these behaviors diverge?

For context: Within `void serialEvent()` we're watching for the string "listen 1", upon which we then call `WiFi.listen()`

This is most easy to replicate when the Photon is flashing green, looking for APs which are not currently in range. Is the system code waiting for lengthy timeouts to occur, attempting to find every stored AP before beginning listening?

This would be a confusing issue for the customer, as WiFi setup mode would be entered through our device's front panel, which then calls `WiFi.listen()` rather than having the customer touch the photon directly. If they press a button and see no behavioral change for >3 mins, that would be quite confusing.

m-mcgowan, 2016-02-04:
How about if you call WiFi.disconnect() first then WiFi.listen()?

pomplesiegel, 2016-02-04:
No difference observed. Just changed the call upon the serial command to 
````cpp
Serial.println("Attempting to enter listening mode");
WiFi.disconnect();
WiFi.listen();
```
this time took 2mins before entering listening.

Does a button press call to another API rather than `WiFi.listen()`, or is it doing so from a different thread?

m-mcgowan, 2016-02-04:
Yes, a button press calls connect_cancel() - I recently added this call also to WiFi.disconnect() but I couldn't recall offhand if this was part of the 0.4.9 release.  I can get this pushed to develop soon if that helps you out of a bind.

pomplesiegel, 2016-02-04:
Sure, not an imminent threat, but would be great to have that fixed for our user testing.

m-mcgowan, 2016-02-05:
@pomplesiegel  we have a fix in place here - https://github.com/spark/firmware/pull/864.

closed 2016-04-03 by technobly