particle-iot/device-os #865: USB connection can put the device in setup mode

m-mcgowan, 2016-02-06:
Currently, setting the line rate of the USB serial connection to 28800 will cause the system to enter "ymodem" mode to perform a system or application update. 

There are a few drawbacks with this:

- The current state of the system on entering ymodem mode is not well defined - the application could have been in the middle of something important. There may be little memory available, or interrupts firing that compete with resources used by the firmware update. 

- Once in ymodem mode, it is not possible to retrieve information about the device, such as the device ID, system firmware revisions. The only option available is to begin a ymodem transfer, or timeout. 

## Proposal

The proposal here is that we change the meaning of the 28800 line rate to put the device in setup mode, and in such a way that existing clients expecting ymodem mode still continue to function. 

The new behavior will be this:

- when the linerate is set to 28800, the system performs a safe mode reset, via `System.enterSafeMode()`

- new flags are set in backup registers that indicate the system should also enter listening mode. These flags are checked as the system attempts to start up and forces the device into listening mode. 

- listening mode currently enters ymodem mode after `f` is received. listening mode is enhanced to allow ymodem to begin immediately if a start packet byte is received, namely `0x01` or `0x02`. These values are not presently used and so are safe to introduce. 

Advantages:

- the application is given a chance to shutdown gracefully rather than being arbitrarily blocked

- the system is in safe mode, a known good state, so no possibility of application interactions causing failure

- setup mode is more expressive than ymodem mode - connected clients can retrieve system firmware details, make configuration changes etc..

- an application or system that was blocked would previously cause a delay before ymodem mode was active. Now the mode will be entered irrespective of what the application or system are doing. (The application is given chance to shutdown the system delays may be introduced there.)

Possible behavior changes:

- a highly responsive application would enter ymodem mode immediately. Under the new scheme, the system still resets, which takes a second or two before the serial line has been connected. (In the USB line rate handler setup there is a 1 second delay. We will investigate why this is there and ideally remove it to speed up reboots.)

andyw-lala, 2016-02-06:
Five questions/comments:

1. What's the state/action _after_ Ymodem transfer completes ?
2. Can/should we support Xmodem too - if I look that far back in my memory,
I seem to recall that both were designed to be automatically detected in a
serial stream (is that what 14400 baud does ?)
3. Is there value to having X/Ymodem listener being active in other serial
modes, not just the 28800 baud case ?
4. What's the value prop of X/Ymodem over other memory write mechanisms, eg
DFU ?
5. It would be nice if all particle cli actions required the same state of
the device. At present some require listening mode, some DFU mode, some
don't care - seems a little messy at present - can X/Ymodem be used to
unify this ?

m-mcgowan, 2016-02-06:
1. After ymodem completes, the device resets to apply the firmware, or returns an error code if the firmware image didn't validate. 

2. 14400 baud triggers DFU mode with our system. I don't see a need for xmodem - all our clients are using our own code presently. 

3. The application firmware can choose to activate ymodem when it's needed. Since this is a blocking function, it should be only active when a file transfer is going to proceed. External devices can trigger ymodem by setting the line rate to 28800. 

4. Ymodem uses USB serial and all our own code which is LGPL licensed. DFU is far more complex, with the only known implementation being dfu-util. This is GPL licensed, which prevents product creators from distributing this without also open sourcing their own firmware. 

5. That is the intent here - that tools can put the device in listening mode at any time so the device can be in any state. Over time, ymodem will be extended to fully replace DFU. The only time DFU will be a hard requirement is if system firmware isn't installed on a device or has become corrupted.

andyw-lala, 2016-02-06:
I *really* like the end game you describe in #5. With that as an outcome, I
think things will be much nicer.

However, I see a conflict with the reset-on-completion behaviour in #1 - is
there a technique we can use/develop to provide an analogue of the ":leave"
behaviour in dfu-util ?

Just thinking out loud here. perhaps have ymodem transfer a payload of some
sort instead of just the raw data, that then gets parsed (and possibly
authenticated to avoid malicious brickage) ?

m-mcgowan, 2016-02-06:
The reset is necessary to replace the system firmware, since that has to be done by the bootloader (so we're not running the firmware as the flash is updated.). What  we will also have is a flag that is a "reset into ymodem again" so that the device stays in ymodem mode so multiple binaries can be flashed in succession without the device attempting to connect to the cloud etc..

andyw-lala, 2016-02-06:
Yeah, but the USB connection will bounce like a superball.

m-mcgowan, 2016-02-06:
Sure, that behavior will be factored into our clients. I don't believe it's a reason for concern (and I don't know of any alternative approach.)

andyw-lala, 2016-02-06:
As I said, you can wrap the payload in something that can contain stuff
like a signature (for authentication), memory target (internal, external,
etc), and a disposition (reset, continue, etc) the wrapper doesn't have to
be complex or result in buffering the entire thing before writing. You
could even create some logical targets instead of just raw addresses and
contents.

m-mcgowan, 2016-02-06:
Understood. Our firmware images are already wrapped in this way, including addresses, checksums, dependencies and other details needed to validate the firmware before applying it, so it's not possible to flash an invalid image by accident. We have to buffer before writing, just like OTA, since we cannot write to the same flash region that is currently running the ymodem code. OTA and OTW transfers share 90% of the code - the primary difference is the transport used. 

We can already flash binaries to the device without specifying any addresses, e.g. `particle flash mydevice system-part2.bin` and the device ensures this goes to the right place. Over the wire updates will be essentially the same.  After each binary is flashed the system resets in order to apply that binary to the target location - this is done in the bootloader so we can be sure the flash regions are not executing code from the region being written.

zsup, 2016-02-07:
I like this approach, although one thing to throw out there is security concerns. Right now, our default security model for the Photon is "physical ownership == virtual ownership" - if you have access to the physical hardware, you can put it in setup mode and from there claim the device to your account. Being able to drop the device into setup mode via a USB serial connection at 28800 opens up another way to make that happen. I don't think it's necessarily a major new security concern - you still have to have physical access to connect to USB - but it's something to consider.

andyw-lala, 2016-02-07:
Can you expand on your concerns, or were you mostly thinking out loud and
wanting to make sure the discussion covered this aspect ?

I think that this does nothing to change the physical access => total
control aspect, but I could be missing something vital, and being only on
the periphery need the dialogue to force me to work through it all.

technobly, 2016-02-07:
Another way to think about it:

- A completely wireless battery powered Photon would need physical access to claim.

- With this proposed change, a USB connected Photon could potentially be claimed remotely through the USB host (e.g., remote desktop, ssh into a RasPi, etc..).  This is not a security issue with the Photon though, but may feel like it.

zsup, 2016-02-07:
Just thinking out loud. I can't think of a real security issue this opens up, but I figured @m-mcgowan @towynlin and @technobly would be quick to identify potential problems so long as it's on their radar

closed 2016-04-03 by technobly