particle-iot/device-os #874: Enhanced support for SMS messages in Electron firmware

sergeuz, 2016-02-12:
Current HAL already has some basic support for SMS messages (though it's not directly exposed as user-accessible API at the moment): https://github.com/spark/firmware/blob/develop/hal/src/electron/modem/mdm_hal.h#L247. The code uses text mode (7-bit ASCII-alike encoding, up to 160 characters) and allows to read incoming messages from SIM card only, which is quite slow and requires polling with periodic cleanup.

What if we add more advanced support for SMS messages? GSM modems have so-called "unsolicited result" mode which can be used to receive SMS messages instantly without using SIM as intermediate storage. Additionally, PDU messaging mode can be used to support long messages, UTF-8 encoding and delivery reports.

Despite the fact that SMS messaging is somewhat obsolete nowadays, I think it would be nice to have additional communication channel available in our platform, that works out of the box, regardless of internet-capable network coverage or connectivity.

jenesaisdiq, 2016-02-12:
Main issue here is that Particle SIMs don't have SMS support- they only have data plans.

sergeuz, 2016-02-12:
@jenesaisdiq Is it possible to receive incoming messages at least, or the functionality is blocked entirely?

zsup, 2016-02-12:
We should test it but I don't believe we can receive inbound messages.
Definitely worth trying though (and we should check what happens on the
billing side when we do)

technobly, 2016-02-15:
@sergeuz my thoughts on SMS were to specifically not create a user api like Cellular.sendSMS() to avoid issues with Particle SIM users not being able to use it.  Instead I'd like to create a Particle Library that exposes the SMS functionality, with an example app that calls out the 3rd party SIM requirements very clearly.  Making it a little more difficult to use and not an official API, will help to mitigate the support issues.

We could also create a whole section of the docs that are dedicated to 3rd party SIM APIs.  I think this is a lower priority though.

To test the incoming SMS capability, I think you'd need to know the phone number of the Particle SIM (which I don't think there is one).  AT+CNUM does this, and is currently in the code.  When I run a Particle SIM I get this:

```
34.582 AT send       9 "AT+CNUM\r\n"
34.626 AT read OK    6 "\r\nOK\r\n"
```

And when I run a 3rd party SMS capable SIM I get (removed my actual number)

```
85.476 AT send       9 "AT+CNUM\r\n"
85.565 AT read  +   32 "\r\n+CNUM: \"\",\"+15555555555\",145\r\n"
85.582 AT read OK    6 "\r\nOK\r\n"
```

sergeuz, 2016-02-15:
@technobly Thanks for the information. It looks like separate library is indeed the best place for this functionality to be implemented.

jenesaisdiq, 2016-02-16:
I'll also chime in to say that we probably don't need to be the ones to write it, if we have any higher priorities at all. Most other firmware improvements will benefit more of our users

technobly, 2016-02-16:
I put it on my list of things I'd like to do in my spare time.  If someone beats me to it though... that's fine, I'll probably still work on it for fun ;-)

jenesaisdiq, 2016-02-16:
Oh, and I should've also written- the "official" SMS path we'll roll out is an integration (twilio/tropo/other) so you can call Particle.text and it basically works as a webhook

m-mcgowan, 2016-04-14:
I'm going to close out this since the consensus is that this won't be part of core firmware, but as a library or integration.

closed 2016-04-14 by m-mcgowan