particle-iot/device-os #879: WiFi.macAddress() is 00:00:00:00:00:00 when using `SYSTEM_MODE(MANUAL)`

nneonneo, 2016-02-19:
When using `SYSTEM_MODE(MANUAL)`, calling `WiFi.macAddress()` within `setup` or the first `loop` iteration results in an all-zeros MAC address being returned.

Test program:

```
#include "spark_wiring.h"
#include "application.h"

SYSTEM_MODE(MANUAL);

void setup() {
	WiFi.on();
	WiFi.connect();

	while(!WiFi.ready())
		;
}

void loop() {
    byte mac[6];
    WiFi.macAddress(mac);
    Serial.printf("WiFi MAC address: %02x:%02x:%02x:%02x:%02x:%02x\n",
        mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
}
```

Invariably, the output from the first `loop` is `WiFi MAC address: 00:00:00:00:00:00`; the output is correct from the second loop onwards.

It can be fixed by using `AUTOMATIC` mode, but for our application this introduces a significant overhead which we don't need. We could also work around it by skipping the first loop but this seems like a bit of a hack.

Is there a function we can call from user (application) code that will refresh the MAC address, preferably callable from `setup`? Alternately, is there a way the firmware can be fixed so that the MAC address is always made available in `MANUAL` mode?

nneonneo, 2016-02-19:
Note: this was tested against the following firmware revision:

```
commit e4e4b975120353052c71a461fc0d194ab2d3633b
Author: Matthew McGowan <mat.mcgowan+github@gmail.com>
Date:   Tue Jan 26 03:20:08 2016 +0100

    updates changelog
```

which was compiled on my machine along with user code in `user/applications`.

m-mcgowan, 2016-02-23:
Thanks for the report. This stems from the way the system calls setup/loop and that setting the MAC address relies upon having the background loop executed.

The proper fix is to have the system internally call `Particle.process()` between `setup()` and `loop()`. It presently doesn't do that, and runs `setup()` and `loop()` in succession, with no background processing in between. 

Until then, as workaround you can call `Particle.process()` in setup after `WiFi.ready()` returns true, which will refresh the MAC address.

m-mcgowan, 2016-04-08:
I think this is just a matter of fetching the ip config when wifi is first activated. I will code and test on Core/Photon.

PR #954 (2016-06-07): makes the MAC address available after `WiFi.on()` in all modes.
fixes #879 

Tested on the Core/P1
---

Doneness:

- [x] Contributor has signed CLA
- [x] Problem and Solution clearly stated
- [x] Code peer reviewed
- N/A  API tests compiled
- [ ] Run unit/integration/application tests on device. Really need the test framework to support reset before each test and setting device modes. 
- [ ] Add documentation
- [ ] Add to CHANGELOG.md after merging (add links to docs and issues)
closed 2016-06-07 by m-mcgowan