particle-iot/device-os #881: [Photon] WiFi.setCredentials() not saving?

technobly, 2016-02-19:
I'm trying to run this app on a Photon v0.4.9 and it stays in a Blinking Blue state.  Would love to see if anyone else can verify this... no need to put your real creds in.

```
#include "application.h"

SYSTEM_MODE(SEMI_AUTOMATIC);

void setup() {
	// for hidden and offline networks on the Photon, the security cipher is also needed
	// Cipher options are WLAN_CIPHER_AES, WLAN_CIPHER_TKIP and WLAN_CIPHER_AES_TKIP
	Serial.begin(9600);
	Serial.printlnf("hasCredentials?: %s", (WiFi.hasCredentials())?"yes":"no");
	delay(2000);
	WiFi.setCredentials("MYSSID", "mypassword", WPA2, WLAN_CIPHER_AES);
	delay(2000);
	WiFiAccessPoint ap[5];
	int found = WiFi.getCredentials(ap, 5);
	Serial.printlnf("Stored Credentials: %d", found);
	for (int i = 0; i < found; i++) {
	    Serial.print("ssid: ");
	    Serial.println(ap[i].ssid);
	    // security is one of WLAN_SEC_UNSEC, WLAN_SEC_WEP, WLAN_SEC_WPA, WLAN_SEC_WPA2
	    Serial.print("security: ");
	    Serial.println(ap[i].security);
	    // cipher is one of WLAN_CIPHER_AES, WLAN_CIPHER_TKIP, WLAN_CIPHER_AES_TKIP
	    Serial.print("cipher: ");
	    Serial.println(ap[i].cipher);
	}
	Serial.printlnf("hasCredentials?: %s", (WiFi.hasCredentials())?"yes":"no");
	delay(2000);
	Particle.connect();
}
```

## Output
```
hasCredentials?: no
Stored Credentials: 0
hasCredentials?: no
```

m-mcgowan, 2016-02-19:
Don't you need to call `WiFi.on()` first?

technobly, 2016-02-20:
Interesting, that worked!  
```
hasCredentials?: no
Stored Credentials: 1
ssid: MYSSID
security: 3
cipher: 1
hasCredentials?: yes
```

However it was unexpected since we don't normally need WiFi.on() for SEMI_AUTOMATIC mode.  After Particle.connect(), WiFi.on() is handled in the background for SEMI_AUTOMATIC.

So the question is... should this be automatically handled for WiFi class function calls in SEMI_AUTOMATIC mode, or should we just document the requirement for WiFi.on() in MANUAL & SEMI_AUTOMATIC modes?

technobly, 2016-02-20:
My vote would be to automatically handle turning on the WiFi module if needed.

m-mcgowan, 2016-02-23:
The `Particle.connect()` call comes after all the WiFi interactions, so WiFi has not been enabled at that point. The intent of the non-automatic modes is that the application developer is given full control over when WiFi is enabled - the system does not enable wifi prior to calling `setup()`. 

I do agree that we either need to document that WiFi.on() is needed or automatically enable the wifi module (and also document that behavior.)  For least user-friction I'm onboard with automatically enabling WiFi when needed.  In rare cases where an application wants to check credentials but not affect WiFi state, they can add a call to

technobly, 2017-02-07:
I think this is mainly a documentation issue, and I'm unsure if automatically turning on WiFi in SEMI_AUTOMATIC mode is a good idea anymore.  There should be a very specific call that takes SEMI_AUTOMATIC mode from a powered off state, to a fully automatically connecting state.  That's either WiFi.connect() and/or Particle.connect().  For the case above, you might want to just turn WiFi.on(), check/set credentials, then turn WiFi.off() for a later time.  If set/check credentials turned power on automatically, then you'd have to know to turn it back off if that was your intension.

technobly, 2017-02-07:
Docs issue added here https://github.com/spark/docs/issues/561, closing.

closed 2017-02-07 by technobly