particle-iot/device-os #888: Electron cannot publish with longer stop mode sleep periods > 25 minutes

bprobbins, 2016-02-26:
I'm finding that electron doesn't publish as expected if sleep mode period is above about 25 minutes.
Is this due to the 23 min ping event setup? Is there a solution?
```
//develop branch with PR#845 fix
#include "application.h"

FuelGauge fuel;
char publishStr[20];
uint32_t start;

void setup() {
  pinMode(D1, INPUT_PULLDOWN);
}//setup()

void loop() {

  sprintf(publishStr, "%.2f %.2f", fuel.getVCell(), fuel.getSoC());
  Particle.publish("e1", publishStr, 60, PRIVATE);

  start = millis();
  while (millis() - start < 1000UL) {Particle.process();}

//will briefly breathe cyan upon awaken but not publish
//unless wakeup < 25 minutes
  System.sleep(D1, RISING, 30 * 60);

  start = millis();
  while (millis() - start < 1000UL) {Particle.process();}
}//loop

```

bprobbins, 2016-02-26:
And then with a slightly different approach I get a stranger still result:
```
//develop branch with PR#845 fix
#include "application.h"

FuelGauge fuel;
char publishStr[20];
uint32_t start;
int sleepInterval = 21; 

void setup() {
  pinMode(D1, INPUT_PULLDOWN);
}//setup()

void loop() {

  sleepInterval++;

  sprintf(publishStr, "%i", sleepInterval);
  Particle.publish("e1", publishStr, 60, PRIVATE);

  start = millis();
  while (millis() - start < 1000UL) {Particle.process();}

//will briefly breathe cyan upon awaken but not publish
//unless wakeup < 25 minutes
  System.sleep(D1, RISING, sleepInterval * 60);

  start = millis();
  while (millis() - start < 1000UL) {Particle.process();}
}//loop
```
Produces: (electron id obscured)
{"data":"22","ttl":"60","published_at":"2016-02-26T05:42:12.289Z","coreid":"560045000b..............","name":"e1"}
{"data":"85CC212725A957348F63008F18EB4EA3A96CF200D1F6B49DFF5C59D7AD2E74C3","ttl":"60","published_at":"2016-02-26T05:42:13.241Z","coreid":"560045000b..............","name":"spark/device/app-hash"}
{"data":"23","ttl":"60","published_at":"2016-02-26T06:04:16.563Z","coreid":"560045000b..............","name":"e1"}
{"data":"24","ttl":"60","published_at":"2016-02-26T06:27:19.010Z","coreid":"560045000b..............","name":"e1"}
{"data":"25","ttl":"60","published_at":"2016-02-26T06:51:20.324Z","coreid":"560045000b..............","name":"e1"}
{"data":"23","ttl":"60","published_at":"2016-02-26T06:51:20.325Z","coreid":"560045000b..............","name":"e1"}
and then stops publishing altogether.

technobly, 2016-02-29:
Hi @bprobbins thanks for filing this issue.  I believe this is due to UDP session move not being included in v0.4.8-rc.6 firmware.  Is that the firmware version you are testing?

We will make sure to test this for an upcoming 0.5.x release.  cc: @m-mcgowan I tagged this issue as 0.5.x.  Please update if you think that is appropriate.

bprobbins, 2016-02-29:
The problem remains in the develop branch, as noted, and i was under the  impression  that develop  had the session move already  merged, so i think the problem  is deeper. Thanks  @technobly for looking  into  this.

technobly, 2016-03-02:
Ahh, thanks!  Didn't see your code comment at first `//develop branch with PR#845 fix`

sbright33, 2016-03-04:
I'm running your code.  When it wakes I get green flashing which means it's handshaking with the tower again.  How can we stop this to save data?

technobly, 2016-03-08:
@sbright33 it should not consume data while blinking green if it's connecting to a tower.  It will consume data as it's blinking cyan as it connects to the particle cloud.  For troubleshooting I recommend posting your questions on the community and/or reaching out to particle support. https://docs.particle.io/support/support-and-fulfillment/menu-base/

bprobbins, 2016-04-03:
Seems fixed now in develop. Thanks!

closed 2016-04-03 by bprobbins