particle-iot/device-os #892: Time.format(Time.local()) give unexpected result.

peterqu, 2016-03-04:
We should update the documentation to include additional clarification about the usage of Time.format.  Time.format(time,) expect the time stamp to be UTC time without offset.

This works correctly.
  Serial.println(Time.format(Time.now(),TIME_FORMAT_ISO8601_FULL));

This would not work correctly, as the Time.format will apply the time zone offset twice.
  Serial.println(Time.format(Time.local(),TIME_FORMAT_ISO8601_FULL));

Test code is here:
void setup() {
  Serial.begin(9600);
  delay(10000);
}

void loop() {
  static String time=""; 
  Time.zone(-8);
  time = Time.hour();
  time += ":";
  time += Time.minute();
  time += ":";
  time += Time.second();
  Serial.println(time.c_str());
  Serial.println(Time.format(Time.now(),TIME_FORMAT_ISO8601_FULL));
  Serial.println(Time.format(Time.local(),TIME_FORMAT_ISO8601_FULL));
  delay(10000);
  
}

m-mcgowan, 2016-04-14:
This is a good point. Given that all the time functions expect UTC time, I'm not sure what the rationale is for `Time.local()`.

@ScruffR Can you help provide rationale for `Time.local()`?

ScruffR, 2016-04-14:
As far as I remember it was in connection with calculating seconds to sleep from now till a specific local time in the future which would require to keep track of the currently set time zone in code since there s no way to read back the time zone from Time object.
When doing calculations on Time.now() and Time.hour() you get inconsistencies since the former uses UTC and the later local time.
Furthermore there is no need for Time.format(Time.local(), "") since Time.format("") does the job with less hassle an confusion. Maybe that would be better to document properly.

technobly, 2017-02-13:
Closing after adding Docs issue here: https://github.com/spark/docs/issues/574

closed 2017-02-13 by technobly

ScruffR, 2018-05-11:
Just documenting it (in some misunderstandable way ;-) ) isn't a solution for all use-cases
Hence I added a new issue #1539 to reopen the case (see also all linked issues)