particle-iot/device-os #893: asynchronous timeout for wlan_connect()

m-mcgowan, 2016-03-05:
We are getting reports that the devices can become locked indefinitely trying to connect to wifi.  I've seen this only sporadically (e.g. 3-4 times in the last year) and have been unable to reproduce consistently. 

We can make a safety net via an asynchronous timer that cancels the connection attempt after 60 seconds.

pomplesiegel, 2016-03-06:
@m-mcgowan, what does this look like from an RGB LED perspective? Does the user code still run, if in mode `SYSTEM_THREAD(ENABLE)`? I'm curious if this could be related to #804

hine0088, 2016-03-09:
We have had a few instances of this in the last week, 

One was with a device that had been functioning normally for weeks with a customer. It lost connection around midnight and then the customer noticed it unresponsive the following morning. The RBG was solid blue

Just had it happen this morning with a device that was reconnecting after first calling WiFi.off(); followed by a WiFi.on(); shortly there afterwards. We have system threading enabled and it actually cut off rendering of a screen image midway through. 

I've gotten it numerous times (2-3) in the last few days while trying to send wifi credentials. I'm not picking out a solid pattern, but it happens when I send a request to the device while connected in soft-ap mode.  e.g. 

    const xmlhttp = new XMLHttpRequest();
    xmlhttp.open('GET', 'http://192.168.0.1/public-key', true);
    xmlhttp.timeout = 4000; // Long timeout needed for scan-ap
    xmlhttp.send();

In these cases, the RGB is solid blue as well.

m-mcgowan, 2016-03-09:
Thanks for the report. This doesn't sound like the same issue - the report here is to have an asynchronous cutoff when WiFi is connecting (Blinking Green LED) while your issue involves setup mode. Of course we want to look into both issues, but in the interests of clarity I'd like to keep them separate. Could you please open another issue or create a post on our community forum, providing as much detail as possible about when the error occurs, version of system firmware, attached hardware and any relevant parts of your application code. Thanks!

hine0088, 2016-03-11:
Absolutely! I missed that the lock state was with blinking green, although in both cases the device was attempting to connect with correct credentials beforehand if that makes any difference. The last few cases appear to be related to the following thread. 

https://github.com/spark/firmware/issues/651#issuecomment-195245409

closed 2016-04-03 by technobly