particle-iot/device-os #894: Sleep lockup issue

jamlamberti, 2016-03-07:
The photon board will lock up if a sleep occurs in a loop and there is not enough time to re-establish the wifi connection. This prevents over-the-air programming with the IDE and requires normal flashing. I would be nice to prevent this or at least warn the user. 

To replicate (tested on latest and develop branches):

```Arduino
int test = D2;
int blink = D7;
int nope=D1;

void setup() {
    pinMode(test,INPUT);
    pinMode(blink,OUTPUT);
    pinMode(nope,OUTPUT);
    
    digitalWrite(blink,HIGH);
    digitalWrite(nope,HIGH);
    delay (1000);
    digitalWrite(blink,LOW);
    digitalWrite(nope,LOW);
}

void loop() {
    digitalWrite(blink,HIGH);
    delay(2000);
    if (digitalRead(test)==HIGH){
        digitalWrite(nope,HIGH);
        System.sleep(10);
    }
    digitalWrite(blink,LOW);
    digitalWrite(nope,LOW);
}
```

Even if the delay is increased to 10 seconds, the device will still lock up. Delaying 30 seconds gives the device enough time to re-connect.

m-mcgowan, 2016-03-07:
I'm not sure what we can do about this without interfering with application behavior. 

A workaround is to ensure the app doesn't put the device to sleep until the device has connected to the cloud, e.g.

```
static bool sleep_when_connected = false;
if (digitalRead(test)==HIGH){
        digitalWrite(nope,HIGH);
        sleep_when_connected = true;
}
if (sleep_when_connected && Particle.connected()) {
   sleep_when_connected = false; 
   System.sleep(10);
}
```

jamlamberti, 2016-03-08:
I agree, but it might be worth having a warning appear if someone tries to verify their code. 

Thanks for the suggestion on checking connection status!

m-mcgowan, 2016-03-09:
I'm afraid that kind of code introspection isn't possible in the general case (it would requite the machine to predict the result of your code, which is often dependent on external factors, such as a button press.) It's not possible to know that the code will not be able to re-establish the wifi connection, or specifically that this is undesirable. In other cases, the application code may truly want to sleep for 10 seconds.

Another alternative is to explicitly connect to the cloud on waking up again:

```
System.sleep(10);
Particle.connect();
waitFor(Particle.connected, 60000);
```

What we can do is that for `AUTOMATIC`/`SEMI_AUTOMATIC` mode, waking up from sleep will wait until the cloud is connected and checks for any OTA messages. (So the system will itself take care of performing the code I posted above.) That would ensure the device is always connected to the cloud before running the application code again on waking up from sleep. How does that sound?

jamlamberti, 2016-03-10:
I think there are 4 ways to address this.

1. Assume the programmer knows best.
 * Pros: the application can function however it wants
 * Cons: OTA messages may not get through and force a DFU update, which the user may not be set up to do

2. What you described above
 * Pros: Ensures OTA messages get through
 * Cons: Application code may execute long enough between sleeps to allow the wifi connection to be re-established. 

3. Check if we are connected to the cloud and check for OTA messages before going to sleep. If we are not connected, delay until we are. Adjust the amount of time we slept for based on how long it takes to re-establish the connection.
 * Pros: OTA messages get through with less change of application behavior
 * Cons: In some cases, this will have the same effect as Option 3 (e.g. Sleep for a small amount of time, execute a small amount of code, sleep for a small amount of time, execute a larger piece of code which allows for cloud connection to be established).

4. Ensure the device can receive OTA messages by periodically ensuring we are connected. If the current time minus the time of the last successful check is greater than some constant (e.g. a minute), ensure we make a connection using either Option 2 or 3.
 * Pros: 

I am not too familiar with the Particle codebase yet so I do not know how feasible some of these options are. Thoughts?

m-mcgowan, 2016-03-10:
These are good thoughts. The reason I prefer nr 2 is because it's the smallest change, and one that aligns strongly with the promise that `AUTOMATIC`/`SEMI_AUTOMATIC` modes presently offer, namely that the system will maintain the cloud connection.   I do feel that not connecting to the cloud on waking from sleep (when Particle.connected() is still true) is an oversight in the implementation rather than a conscious decision. 

These modes provide a guarantee to the application that the system will maintain the cloud connection in the background. It does this before calling setup()/loop() and it makes sense to do this also when waking from sleep. 

AUTOMATIC mode is very much the converse of 1. If the programmer knows best then they can chose MANUAL mode and manually re-connect to the cloud, or disconnect the cloud before going to sleep so that the cloud remains disconnected on waking up. 

I'll add this for the 0.5.x release. Thanks for raising the issue!

m-mcgowan, 2016-03-14:
The sleep API still confuses even me - `System.sleep(10)` only sleeps the wifi network, and not the entire system, so the system will still be running the main loop. Is this what you wanted?

jamlamberti, 2016-03-14:
My understanding is that a ```System.sleep(10);``` sleeps the application code as well as the wifi module. When the sleep ends (either by timing out or by an interrupt), the main loop will continue executing and the wifi module will start trying to reconnect and check for OTA messages.

If you have the system go back to sleep before you can send it an OTA message, then you have to resort to building the firmware, and flashing the board over DFU.

m-mcgowan, 2016-03-14:
I'm afraid that's not correct. You can see yourself from looking at the source that sleep 10 only disables the wifi network and does not stop the application from running. 

```
System.sleep(10);
```

Is an inline function that expands to

```
    inline static void sleep(long seconds) { sleep(SLEEP_MODE_WLAN, seconds); }
```

jamlamberti, 2016-03-14:
Interesting. I observed the behaviour of the application stopping when using interrupts. But I guess this is due to entering stop mode. I should have read the doc in more depth. 

```C
void system_sleep_pin(uint16_t wakeUpPin, uint16_t edgeTriggerMode, long seconds, uint32_t param, void* reserved)
{
    network_suspend();
    LED_Off(LED_RGB);
    HAL_Core_Enter_Stop_Mode(wakeUpPin, edgeTriggerMode, seconds);
    network_resume();
}
```

So I guess the issue with ```System.sleep(x)``` is if you have less than ```x``` seconds of execution before going back to sleep, the device will never have a wifi connection.

closed 2016-04-03 by technobly