particle-iot/device-os #896: TCPClient on Electron not receiving all of the data for small files

technobly, 2016-03-09:
As reported here in the community:
https://community.particle.io/t/electron-tcpclient-weirdness/20771

Test code, press 'y' to read an 8 byte file.  Typically the header is not even completely printed in the serial console, let alone the 8 bytes of data.  That said, when looking at the logs we do see that all of the data is read, but the system is doing some strange things.  See logs below:

## Code
```cpp
#include "application.h"

// ALL_LEVEL, TRACE_LEVEL, DEBUG_LEVEL, INFO_LEVEL, WARN_LEVEL, ERROR_LEVEL, PANIC_LEVEL, NO_LOG_LEVEL
SerialDebugOutput debugOutput(9600, ALL_LEVEL);

SYSTEM_MODE(AUTOMATIC);

void setup()
{
    Serial.begin( 9600 );
}

/* This function loops forever --------------------------------------------*/
void loop()
{
    if (Serial.available() > 0)
    {
        char c = Serial.read();
        Serial.printf("Hey, you said \'%c\', so I'm gunna: ", c);
        if (c == 't') {
            uint32_t startTime = millis();
            uint32_t endTime = 0;
            Serial.printlnf("Read a text file\r\n[ START TIME ]: %.2f seconds", startTime/1000.0);
            TCPClient client;
            char host[] = "www.textfiles.com";

            Serial.println("connecting...");

            if (client.connect(host, 80))
            {
                Serial.println("connected");
                /* SMALL FILE */
                client.println("GET /100/hack7.txt HTTP/1.0"); // 3988 bytes
                /* LARGE FILE */
                // client.println("GET /100/phrack.29.phk HTTP/1.0"); // 235777 bytes
                client.println("Host: www.textfiles.com");
                client.println("Content-Length: 0");
                client.println();
            }
            else
            {
                Serial.println("connection failed");
            }

            while (client.available() || client.connected())
            {
              if (client.available())
              {
                char c = client.read();
                Serial.print(c);
              }

              if (!client.connected())
              {
                Serial.println();
                Serial.println("disconnecting.");
                client.stop();
              }
            }
            endTime = millis();
            Serial.printlnf("[ END / TOTAL TIME ]: %.2f / %.2f seconds", endTime/1000.0, ((endTime-startTime)/1000.0));
        }
        else if (c == 'y') {
[... 133 lines cut ...]
    87.914 AT read OK    6 "\r\nOK\r\n"
0000087918:DEBUG: virtual int TCPClient::available() (145):recv(=128)
HTTP/1.1 200 OK
Date: Wed, 09 Mar 2016 21:47:27 GMT
Server: Apache
Last-Modified: Wed, 09 Mar 2016 21:19:09 GMT
ETag: "d0a9a    87.939 AT read  +   18 "\r\n+UUSORD: 1,116\r\n"
Socket 1: handle 1 has 116 bytes pending
socketRecv: _cbUSORD
    87.961 AT send      16 "AT+USORD=1,128\r\n"
    87.986 AT read  +  134 "\r\n+USORD: 1,116,\"d-8-52da43fac4c1a\"\r\nAccept-Ranges: bytes\r\nContent-Length: 8\r\nConnection: close\r\nContent-Type: text/plain\r\n\r\ndeadbeef\""
    88.015 AT read OK    6 "\r\nOK\r\n"
socketRecv: _cbUSORD
    88.021 AT send      15 "AT+USORD=1,12\r\n"
    88.236 AT read ERR  37 "\r\n+CME ERROR: operation not allowed\r\n"
    88.243 AT read  +   14 "\r\n+UUSOCL: 1\r\n"
Socket 1: handle 1 closed by remote host
0000088263:DEBUG: virtual uint8_t TCPClient::connected() (209):caling Stop No more Data and in CLOSE_WAIT
0000088273:DEBUG: virtual void TCPClient::stop() (192):_sock 1 closesocket
socketFree(1)

disconnecting.
0000088285:DEBUG: virtual void TCPClient::stop() (192):_sock -1 closesocket
[ END / TOTAL TIME ]: 88.29 / 8.00 seconds
```

m-mcgowan, 2016-03-09:
Most socket APIs have a read method that lets you pass in a buffer and the API will read as much as is available, up to that amount, but potentially less. But here the modem is expecting the exact amount of data to be retrieved, so before the _chUSORD call, the code will need to check how much data is available and limit the amount read to that.

m-mcgowan, 2016-03-09:
The proposed fix above is untested - I thought it would be quicker that you did it since you have the test fixture already setup.

technobly, 2016-03-09:
@m-mcgowan your fix is working well!  I didn't remember there was a while(len) loop in the socketRecv() function... which is where the extra remainders of reads were coming from (i.e. 128-116=12 bytes from above logs)... resulting in an error (-1) being returned instead of the byte count (cnt).

asterix555, 2016-03-12:
Hi,

I merged the commit, compiled a firmware, modified build/version.mk and flashed system-part1.bin, system-part2.bin and test.bin and verified using particle serial inspect. The version number increased, but the result isn't as expected,..

[ START TIME ]: 91.74 seconds
connecting...
connected
HTTP/1.1 200 OK
Date: Sat, 12 Mar 2016 11:57:50 GMT
Server: Apache
Last-Modified: Wed, 09 Mar 2016 21:19:09 GMT
ETag: "d0a9a[ END / TOTAL TIME ]: 95.92 / 4.18 seconds

technobly, 2016-03-17:
@asterix555 you should only need to:
```
git fetch
git checkout feature/electron_tcp_socket
```
...to test this fix.  Can you try again and make sure you see the fix in the last commit before compiling?
```
commit 32a436177f022101eb334b4e3abcdc45f7500660
    suggested fix for #896 (untested)
```

asterix555, 2016-03-17:
hi,
the commit is in my code. I did a make PLATFORM=electron clean before and increased version.mk to 0.4.9e and verified, that this firmware is correctly flashed. After flashing system-part1&2.bin 3 times in listening mode it increased.

I also tried to flash OTA, same result.

    $ git status
    On branch feature/electron_tcp_socket
    Your branch is up-to-date with 'origin/feature/electron_tcp_socket'.

Check Firmware: 

    system firmware version: 0.4.9e

Result:

    Hey, you said 'y', so I'm gunna: Read a text file
    [ START TIME ]: 7.58 seconds
    connecting...
    connected
    HTTP/1.1 200 OK
    Date: Thu, 17 Mar 2016 05:54:29 GMT
    Server: Apache
    Last-Modified: Wed, 09 Mar 2016 21:19:09 GMT
    ETag: "d0a9a[ END / TOTAL TIME ]: 14.35 / 6.77 seconds

technobly, 2016-03-17:
Thanks for the quick reply @asterix555 !  So were you just modifying version.mk to make sure the flash memory was updated?

I'll test this again in the morning (1AM now)... it was working fine for me when I initially tested.  What type of Electron do you have? U260/U270/G350?

asterix555, 2016-03-17:
Hi,
it's the G350. Thank you so far and good night!

entone, 2016-03-17:
using the feature/electron_tcp_socket branch fixed my issue. thanks!

technobly, 2016-03-18:
Confirmed again it's working on a U260, but just now trying on a G350 I'm getting the old response where it doesn't show the full message even though it's definitely receiving it.  Just like @asterix555 's output... will investigate.

asterix555, 2016-03-25:
Hi, let me know if you have any code for testing,.. regards

entone, 2016-03-25:
I'm using a modified version of HttpClient, https://github.com/NationalAssociationOfRealtors/SensorNode/blob/electron/libraries/HttpClient/HttpClient.cpp 

and the calling code is here https://github.com/NationalAssociationOfRealtors/SensorNode/blob/electron/sensor_node.cpp#L94

technobly, 2016-03-25:
@entone are you using a G350 or U260/U270?

entone, 2016-03-25:
U260

technobly, 2016-03-25:
Ok cool, thanks @entone.  So far @asterix555 I have no fix for G350's, and it something I will be looking at fixing in the near future.

doom369, 2016-03-29:
@technobly hello, did you have a chance to look through the issue?

technobly, 2016-03-29:
Not yet @doom369, but it is on my list of things to look into ;-)  I'll update here when I have more info.

doom369, 2016-03-29:
Ok, Thanks.

vshymanskyy, 2016-03-30:
@technobly , Blynk is probably blocked by this issue: https://github.com/blynkkk/blynk-library/issues/104
I have SARA-U270...

technobly, 2016-04-14:
@asterix555 @doom369 @vshymanskyy please give https://github.com/spark/firmware/pull/962 a test on your G350's.  Should be working for you now with large and small files.

asterix555, 2016-04-14:
Thank you.
I'm abroad now and can test it only after end of april...

yarrumretep, 2016-04-14:
@technobly - when do you think these changes will be bundled into a formal release and/or merged to master?

closed 2016-04-14 by m-mcgowan

technobly, 2016-04-14:
@yarrumretep working on it now... check Build tomorrow (Friday) :)

technobly, 2016-04-16:
Build rollout pushed to early next week... hopefully Monday.  Binaries are out here though for system part updates: https://github.com/spark/firmware/releases/tag/v0.5.0-rc.2

commit 44e6c1b7 (2016-04-14): fixes #896 for G350

U260 and G350 both were recv a +UUSOCL URC for a remotely closed socket. The U260 would receive this later in time and therefor did not see any issue.  The G350 would recv the URC earlier which caused the socketReadable() routine to throw an unexpected error by setting OK to false.  This was keeping the `cnt` from being returned for the last bytes received.  When not `readable` now, we ensure `len` is set to 0 and OK is set to `true`.