particle-iot/device-os #897: [Photon] enters breathing blue state

kennethlimcp, 2016-03-10:
For some reason `while(!Particle.connected()` results in a breathing blue state:

```
SYSTEM_MODE(SEMI_AUTOMATIC);

void setup(){
    WiFi.on();
    Particle.connect();
    while(!Particle.connected());
}
```

m-mcgowan, 2016-03-10:
More details please! Here's some things to thing about:

- how reproducable is this? 
- do you have wifi creds configured? 
- what happens if you add WiFi.connect() before Particle.connect()?
- have you found any workaround?

kennethlimcp, 2016-03-10:
- 100% reproducable
- Yup wifi has been configured. Placing the same photon in Safe mode and reflashing without that line works
- tried moving that line into `loop()` and the same behavior happens
- haven't found a workaround but it seems like calling `Particle.connected()` rapidly with `while(!Particle.connected()` results this issue.
- I don't have a debugger hooked up to see what's going on yet.

m-mcgowan, 2016-03-10:
Cool. Thanks for the info and thanks for reporting. I'll see if I can reproduce it when I'm finished with my current stack of things.

kennethlimcp, 2016-03-10:
Adding a small delay, you will notice a brief solid blue before flashing green and cyan..

```
SYSTEM_MODE(SEMI_AUTOMATIC);

void setup(){
    WiFi.on();
    Particle.connect();
    while(!Particle.connected())
        delay(2);
}
```

m-mcgowan, 2016-03-10:
oh, in my mind there was a Particle.process() call.  Breathing blue is correct for WiFi being on but not connected to wifi.

m-mcgowan, 2016-03-14:
Closing - this is expected behavior.

closed 2016-03-14 by m-mcgowan

kennethlimcp, 2016-03-14:
So breathing blue forever without a delay is expected with the above code?

m-mcgowan, 2016-03-14:
Which above code there's two examples. In the first example there's an unending loop with no call to the system to initiate the connection, so breathing blue indefinitely is the outcome. The 2nd example, there is a call to delay() so the device can connect to the cloud.