particle-iot/device-os #899: program-cloud makefile target only working for monolithic builds (Spark Core namely)

technobly, 2016-03-11:
Works for Core (monolithic build):
`firmware/main $ make clean all PLATFORM_ID=0 -s program-cloud SPARK_CORE_ID=1234123412341234 SPARK_ACCESS_TOKEN=2345234523452345`

Does not work for Photon (modular build):
`firmware/main $ make clean all PLATFORM_ID=6 -s program-cloud SPARK_CORE_ID=1234123412341234 SPARK_ACCESS_TOKEN=2345234523452345`

Fails with monolithic Photon build (file too large):
`firmware/main $ make clean all PLATFORM_ID=6 -s program-cloud SPARK_CORE_ID=1234123412341234 SPARK_ACCESS_TOKEN=2345234523452345 MODULAR=n`

CLOUD_FLASH_URL not visible to modular builds:
https://github.com/spark/firmware/blob/aefb3342ed50314e502fc792f673af7a74f536f9/main/import.mk

Should we make this available for system parts as well?  Would seem necessary for modular... and hopefully would work in modules/ and main/ directories.

m-mcgowan, 2016-03-11:
which commit did you run this against?

technobly, 2016-03-11:
latest develop

technobly, 2016-03-11:
I'll keep this open until we ship 0.5.x

closed 2016-03-31 by technobly