particle-iot/device-os #904: Photon stuck on pulsing green

airbornemint, 2016-03-12:
I have a Photon that is stuck in a weird state. Here's what I know:

 * Firmware version is 0.4.9
 * System thread is enabled
 * Application thread is repeatedly running loop() as expected
 * Status LED has been breathing green for a long time (1+ hour)
 * My outbound network connection is working fine
 * WiFi.ready() returns true
 * Particle.connected() returns false
 * Calling WiFi.listen() does *not* cause the status LED to flash blue
 * Calling WiFi.listen() does *not* activate the Photon-NNNN WiFi network

That's all I was able to conclude from knowing which code paths application code is going through; I suspect that when I eventually push the reset button, the Photon will get out its weird state and acquire cloud connection.

m-mcgowan, 2016-03-12:
What system mode are you using?

airbornemint, 2016-03-12:
Automatic.

m-mcgowan, 2016-03-12:
Thanks. Does your code call Particle.disconnect() at any time?

airbornemint, 2016-03-12:
Not directly, but it does call WiFi.listen(true) and (after interacting with the user) WiFi.listen(false). I imagine that WiFi.listen(true) implicitly calls Particle.disconnect.

I think, but I am not sure, that I got into this state after a call to WiFi.listen(false), at which time my Photon reconnected to WiFi, but got stuck there for some reason.

m-mcgowan, 2016-03-18:
Just to be clear - does pulsing green mean flashing green (on and off repeatedly) or does it mean breathing green - fading in and out slowly?

airbornemint, 2016-03-18:
Breathing.

m-mcgowan, 2016-03-18:
If you can provide steps to reproduce this that would be a great help. I've looked though the code and don't see that the system is changing the "connect to the cloud" flag. The only place is if

```
WiFi.off(true)
```

Is called, which disconnects the device from the cloud, requiring Particle.connect() to be called to reassert the connection flag.

airbornemint, 2016-03-18:
Yeah, it's extremely unlikely I'll be able to do that.

On this particular day, I rebuilt my device on a custom PCB (instead of a breadboard), and was putting it through its paces. 

I have a code path (which I was repeatedly exercising) that calls WiFi.listen(true). (Basically, there is a button the user can push to force my device into listening mode, at which point they can connect to it using the Particle mobile app and give it new WiFi credentials.)

I was testing this because there was a bug in 0.4.8 which prevented WiFi.listen(false) from working after WiFi.listen(true) had been called, and I wanted to make sure that it is fixed. (Basically, there is a second button the user can push that means "never mind, I don't want to set up WiFi after all".)

During one of the dozens of times that I exercised the WiFi.listen(true) → WiFi.listen(false) code path, I got the device into a state where — as described above — it was breathing green (so, supposedly connecting to WiFi but not to the cloud). 

At this point (and for no good reason :-) ), I tried WiFi.listen(true) again and noticed that the device was not only unable to connect to cloud, but also was refusing to enter AP mode (status light never flashed blue). Interestingly, though, I was able to otherwise interact with my app, and was running through the WiFi.listen(true) → WiFi.listen(false) code path).

At that point I concluded that this isn't right, and filed a bug.

Since this was a rare occurrence, I am happy to let you know if it repeats itself, but I am not sure if I am going to be able to do anything more than tell you "oh hey it happened again", as most of the time my device enters and exists AP mode, and then reconnects to the cloud, without any problems, and as far as I know I didn't do anything special in the failure case (leading me to suspect a timing-dependent bug). 

I guess the only suggestion that I have is that I am 100% sure that the my code was calling WiFi.listen() and also 100% sure that the status light was not changing to flashing blue, so if you are going to go digging through code to try to figure out what's going on, maybe focusing on those two pieces of information would be productive? That seems like the most aberrant part of the whole puzzle.

m-mcgowan, 2016-03-18:
Another reason for the device to fallback to breathing green is if the system cannot maintain the cloud connection - such as if the system thread is blocked (and it sounds like this is the case.)

The system thread will block for a few seconds when attempting to post an event to the application queue when the queue is full. This can happen when application is not processing application events (such as when the application is stuck in a tight loop without calling `Particle.process()`.

airbornemint, 2016-03-18:
I am using a protothreads library (basically, it's a [macro-based stackless thread library](http://dunkels.com/adam/pt/index.html)), so there is no way I would have been able to interact with my device's UI at all if I was stuck in a tight loop. I know for sure that my loop() was returning, but that's an interesting direction of thought. If there anything else that might cause that queue to fill up? Could there be a leak related to WiFi.listen being called many times over that could fill up the queue?

markterrill, 2016-03-20:
Hi,
I bumped into this today, was actually coming in here to post the bug.

My issue was on semi_automatic mode and using an isr to trigger a timeout on connect(). Easily triggered via:
```
WiFi.disconnect(); // to enter listen mode more quickly
WiFi.listen(true);
```

More of the code is over in the forum, on a thread you've been talking on: https://community.particle.io/t/core-code-wont-compile-wlan-serial-config-done-not-declared-in-this-scope/15603/5

It took me a while watching detailed debugging to figure out the softap *should* be on and to scan for it, and it was there. I ended up coding a manual rgb.control to flick the led to orange when it was in 'my' listen mode. It even looks good.

Cheers!

Now I'm realising that the nice orange causes issues with my manual and on screen instructions. Dear lazy web, what RGB is the listen mode blue?

pomplesiegel, 2016-03-22:
We quickly ran into a photon breathing green when experimenting with `SYSTEM_MODE(AUTOMATIC)` today as well. We do not at any point call `Particle.disconnect()`

pomplesiegel, 2016-03-22:
This occurred using the tip of `develop`, as of today.

pomplesiegel, 2016-03-24:
Seeing this Breathing Green + stopped user application (and our sw timer processes) today, using the tip of `develop` as of yesterday, even when in `SYSTEM_MODE(MANUAL)`

m-mcgowan, 2016-04-08:
Any tips how we can reproduce this?

pomplesiegel, 2016-04-08:
This may be the same issue as #804, given the similar outward symptoms. We just shipped off a kit to @monkbroc of programmed HW to reproduce that issue.

airbornemint, 2016-04-08:
For what it's worth: the problem I reported here involves a Photon stuck in breathing green with application thread running normally. 

(Whereas #804 might be about a crash in which app thread is no longer running.)

pomplesiegel, 2016-04-10:
@airbornemint, I Totally agree, and this is a very important distinction. I just figured there could be some connection or perhaps this is a pre-cursor to a more major crash.

markterrill, 2016-04-11:
Hi, my experience is that yes, application thread keeps running, interrupts keep running, there is nothing wrong, **except the RGB does not change color**

pomplesiegel, 2016-05-02:
Update: Today we encountered this same issue: user application still running but cloud connection lost and RGB LED in breathing green indefinitely. This is against the tip of `develop`.

pomplesiegel, 2016-05-02:
This is easily reproducible when running our app and calling function `call`s and `get`s repeatedly using a bash script. I'll try and isolate the issue on a binary other than our application, but i'm glad we're both seeing it.

pomplesiegel, 2016-05-12:
@m-mcgowan, we now know how to easily reproduce this issue. We have seen the issue occur naturally with our own program by simply allowing it to run w/ calls/gets occurring for about a day, but we can now reproduce this using just the simple program below, compiled against the tip of `develop` + PR #984 

````cpp
#include "application.h"

SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

int DoStuff(String unused)
{
  Serial.println("DoStuff called");
  System.reset();
  return 1;
}

void setup()
{
  Serial.begin(9600);

  WiFi.on();
  WiFi.connect();
  Particle.connect();

  Particle.function("DoStuff", DoStuff);

  Serial.println("Example program with resets upon function call");
}

void loop()
{
  Particle.process();
}
````

We simply made a bash script to repeatedly call `DoStuff()` OTA, which then resets the Photon and eventually brings it to a crashed Green RGB LED state. FYI, the RGB LED was totally stalled on green in this case - no longer breathing. 

Can you please give this a shot on your end? We'd love to end this "Breathing Green" problem, as it's our largest risk System FW issue before we begin shipping.

m-mcgowan, 2016-05-12:
That's great news! How long does it take for the problem to appear?

pomplesiegel, 2016-05-12:
I restarted the test w/ this new code at  about noon and noticed the problem now, so it's < 12 hours to reproduce. We're likely attempting to reset OTA every 30s - i'd have to check the script again.

pomplesiegel, 2016-06-01:
Still seeing this (including when I walked back to my desk) when running a full application (rather than just a simple test script), but unable to provide greater information, as we use the multiplexed JTAG pins for critical operations within our program. This occurs with system firmware 0.5.1. 

Other users are seeing this breathing green as well. **How can we work to resolve this?** It is a very real issue and could occur to any unit in the field, making it unreachable.

We have seen the breathing green LED behavior in a few flavors
1. User thread running but sw timers stalled (hack-fixed by hardware watchdog)
2. User thread and sw timers both stalled (hack-fixed by hardware watchdog)
3. User thread and sw timers both running (potentially fixed by turning WiFi chip off, then on and reconnecting)

All are indicated through the breathing green LED.

monkbroc, 2016-06-02:
We have enough information and reports to know that there are several scenarios where the LED can be stuck breathing green.

Mat highlighted one cause above: if the system thread is blocked, either waiting for a RTOS event without a timeout or in an infinite loop.

So far the firmware team hasn't been able to make the issue appear in a Photon.

Anybody who sees the issue on a regular basis can follow these steps to capture useful information about the state of the system.

@pomplesiegel I know you said you can't get access to pins D3 to D7. I'm including these steps for anybody else who is seeing the issue and may be able to send us this information.

## Steps to capture information about stuck Photon

Make sure the application doesn't use D3 to D7 since those are JTAG pins.

Connect the device to a programmer shield.

Recompile the application without modules to get only one ELF file.
```
# in firmware/main folder, with your app code in user/applications/my_app
make all program-dfu PLATFORM=photon MODULAR=n USE_SWD_JTAG=y APP=my_app
```
Run OpenOCD with RTOS support
```
openocd -f interface/ftdi/particle-ftdi.cfg -f target/stm32f2x -c "gdb_port 3333" -c "\$_TARGETNAME configure -rtos FreeRTOS"
```

Open GDB with the ELF file
```
arm-none-eabi-gdb -ex "target remote localhost:3333" build/target/main/platform-6/my_app.elf
```

Wait for the breathing green to happen then capture a stack trace `bt` for all threads: `thread apply bt`

Also capture general info about which thread are running with `info threads`

DaliborFarny, 2016-06-06:
Hello, we are experiencing the same issue - Photon gets stuck, the LED is breating or blinking cyan, the user code is not running.

- system firmware 0.5.1
- SYSTEM_THREAD(ENABLED);
- system mode automatic
- we use local cloud installation on our server in Europe

We decided for SMT mounting of the Photons (proper copper keepout on both layers) and are experiencing also DHCP issues described here: https://community.particle.io/t/photon-no-dhcp-server-in-listening-mode/18727
Maybe the issues are somehow connected..

We deployed around 15 products (clocks) to customers, 4 clocks show this issue (Photon freezing). I will try to capture the stack trace using the JTAG and post it here.
[image]
[image]

zsup, 2016-06-06:
@DaliborFarny sorry to go off-topic for a moment, but I would _love_ to buy one of those nixie clocks. Are they for sale?

pomplesiegel, 2016-06-06:
Hi @DaliborFarny, the issue we're describing here is mainly breathing GREEN. Have you seen it crash or stop running in the expected manner in that color? 

We haven't seen any cyan crashes since the WICED issue #877 was fixed.

DaliborFarny, 2016-06-06:
@zsup of course - please send me email to dalibor[at]farny.cz, I will send you details. thanks!

DaliborFarny, 2016-06-06:
@pomplesiegel 
I had expected the patch for WICED issue #877 would solve those crashes, but it didnt. We still get the crashes with 0.5.1 firmware. The LED is stuck at blinking or breathing CYAN. Most customers hadnt experienced this behaviour at all, but one of my customers who has three clocks at office (connected via LTE - MiFi) had them crashing all. One clock has photon on headers, this crashed only twice in few months, two clocks with SMT mounted photons crash several times per week.

m-mcgowan, 2016-06-06:
@DaliborFarny  Just for reference - the crashes we fixed in #877 were a complete system stall - no LED activity (LED would be constant on or off.) If the LED is still blinking/breathing then the system hasn't crashed and you're seeing a different issue, such as the system failing to reconnect to the cloud in automatic mode.

DaliborFarny, 2016-06-06:
@m-mcgowan thank you for response, the crashes I have prevents the user code from running. I have SYSTEM_THREAD(ENABLED); so I would expect the system thread (which might be waiting for reconnection) not blocking the user code?

m-mcgowan, 2016-06-06:
Right, that's how it is usually. But if the user code posts so many requests to the system that the system queue is full (and the system thread is itself blocked from pulling events off the queue, such as on a never-ending network operation) then the app will block until the queue has room for the new event.  

Perhaps one way to track this down is to introduce additional logging to that we can get a clearer picture of exactly what the system is doing and why it's not emptying the queue. I'll make a PR with additional logging and instructions for how to enable this.  Thanks for bearing with us and helping track down the issue.

DaliborFarny, 2016-06-06:
@m-mcgowan great! I will make whatever I can to figure out this issue, it is the highest priority for us. Thanks for your time!

pomplesiegel, 2016-06-06:
@m-mcgowan, Would the scenario you're describing explain just @DaliborFarny's cyan issue, or potentially the breathing green issue (this thread is named for) as well?

m-mcgowan, 2016-06-06:
Hi @pomplesiegel!  Yes, it can explain all cases where the system doesn't make any progress (which eventually may cause the application to not make progress either) - this covers all the remaining cases discussed here (breathing green + breathing/blinking cyan).  

I suspect a bug in WICED, LwIP or some other layer below the main system task that is causing the system to hang on a network operation. 

In the apps that are failing, do they use network connections in addition to the cloud connection?

airbornemint, 2016-06-06:
Mine doesn't do any networking other than the Particle cloud networking performed by the Particle firmware.

m-mcgowan, 2016-06-06:
Great! Any further details you can share regarding what triggers the problem? For example, Is it related to intermittent wi-fi connectivity?

DaliborFarny, 2016-06-06:
Most of the time, my app does only the Particle cloud networking. The only other networking is done when user changes settings of the clock (time zone, display settings etc.) using our website. The website calls function registered to the cloud.

airbornemint, 2016-06-06:
Unfortunately, I don't have any information beyond what I originally reported upthread, the short version of which is "Photon 0.4.9 flashing green after calls to Particle.disonnect() and Particle.connect(), not reproducible"

DaliborFarny, 2016-06-06:
@m-mcgowan I dont know what triggers the problem, I was not able to track it down to any specific cause. For example, one clock was running well for two weeks, then got stuck a couple of times in one day. I see this behaviour in both 0.5.0 and 0.5.1.

DaliborFarny, 2016-06-07:
@m-mcgowan Hi, I am ready to test the mentioned logging.. Another clue for solving the issue: two clocks that got stuck at least once per day at customer's office in California were sent back to me. I was not able to reproduce this "freezing" issue in a week. I am closer to the cloud (local cloud at server in Europe).

monkbroc, 2016-06-07:
@DaliborFarny If there's any way you can connect D3 to D7 to JTAG on a programmer shield and [follow the instructions above to get a stack trace](https://github.com/spark/firmware/issues/904#issuecomment-223349018) when the problem occurs, that would be fabulous!

DaliborFarny, 2016-06-07:
@monkbroc okay, I need to get the programmer shield first, then I will give it a try. Our app uses D3-D6, I will reconfigure the pins for the test.

pomplesiegel, 2016-06-08:
@m-mcgowan, sounds like the logging would be quite helpful for those of us who need to use the JTAG pins (D3-D7) for our program. When can we expect that PR?

DaliborFarny, 2016-06-09:
Hi @m-mcgowan, I am waiting for the PR as well. I will try also the JTAG approach, but this will take some time to set up.

m-mcgowan, 2016-06-09:
I'm pretty snowed under now, I'll try to get to it early next week. You can in the meantime try building the debug version of firmware and adding logging to your app. Details are in https://github.com/spark/firmware/blob/develop/docs/debugging.md

DaliborFarny, 2016-06-11:
Today I got the photon stuck three times, once blinking cyan, once blinking green and once BREATHING GREEN.

DaliborFarny, 2016-06-13:
@m-mcgowan, I built and flashed the debug enabled firmware to the Photon. Also the Programmer shield is on the way to me, will be here in a couple of days. As I mentioned, we are using a local cloud to connect the Photons to. I set up another instance of local cloud, this time in Los Angeles (current one is in Europe), the issue persists.
I attach the output from the local cloud script, it is just before the Photon got stuck, the _82 session is the last.

> Connection from: 46.167.237.160, connId: 77
on ready { coreID: '3f003c000447343232363230',
  ip: '46.167.237.160',
  product_id: 6,
  firmware_version: 65535,
  cache_key: '_76' }
Core online!
1: Core disconnected: socket error Error: read ECONNRESET { coreID: '3f003c000447343232363230',
  cache_key: '_76',
  duration: 18.035 }
Session ended for _76
Connection from: 46.167.237.160, connId: 78
on ready { coreID: '3f003c000447343232363230',
  ip: '46.167.237.160',
  product_id: 6,
  firmware_version: 65535,
  cache_key: '_77' }
Core online!
onSocketData called, but no data sent.
1: Core disconnected: socket close false { coreID: '3f003c000447343232363230',
  cache_key: '_77',
  duration: 28.866 }
Session ended for _77
Connection from: 46.167.237.160, connId: 79
on ready { coreID: '3f003c000447343232363230',
  ip: '46.167.237.160',
  product_id: 6,
  firmware_version: 65535,
  cache_key: '_78' }
Core online!
1: Core disconnected: socket error Error: read ECONNRESET { coreID: '3f003c000447343232363230',
  cache_key: '_78',
  duration: 27.545 }
Session ended for _78
Connection from: 46.167.237.160, connId: 80
on ready { coreID: '3f003c000447343232363230',
  ip: '46.167.237.160',
  product_id: 6,
  firmware_version: 65535,
  cache_key: '_79' }
Core online!
onSocketData called, but no data sent.
1: Core disconnected: socket close false { coreID: '3f003c000447343232363230',
  cache_key: '_79',
  duration: 28.942 }
Session ended for _79
Connection from: 46.167.237.160, connId: 81
on ready { coreID: '3f003c000447343232363230',
  ip: '46.167.237.160',
  product_id: 6,
  firmware_version: 65535,
  cache_key: '_80' }
Core online!
1: Core disconnected: socket error Error: read ECONNRESET { coreID: '3f003c000447343232363230',
  cache_key: '_80',
  duration: 56.97 }
Session ended for _80
Connection from: 46.167.237.160, connId: 82
on ready { coreID: '3f003c000447343232363230',
  ip: '46.167.237.160',
  product_id: 6,
  firmware_version: 65535,
  cache_key: '_81' }
Core online!
1: Core disconnected: socket error Error: read ECONNRESET { coreID: '3f003c000447343232363230',
  cache_key: '_81',
  duration: 16.853 }
Session ended for _81
Connection from: 46.167.237.160, connId: 83
on ready { coreID: '3f003c000447343232363230',
  ip: '46.167.237.160',
  product_id: 6,
  firmware_version: 65535,
  cache_key: '_82' }
Core online!
1: Core disconnected: socket error Error: read ETIMEDOUT { coreID: '3f003c000447343232363230',
  cache_key: '_82',
  duration: 765.687 }
Session ended for _82

pomplesiegel, 2016-06-14:
Hi @m-mcgowan and @monkbroc, we just reproduced the breathing green issue on a board running w/ the DEBUG build of our full application + the tip of develop + `SerialLogHandler logHandler(LOG_LEVEL_ALL)`. We captured the serial log over about 12 hours. 

Here were the order of events 

A. Everything behaving normally, repeatedly logging things like this
````
0026899568 [hal] socket_hal.cpp:792, read_packet_and_dispose(): TRACE: Socket 0 receive bytes 2 of 18
0026899568 [hal] socket_hal.cpp:792, read_packet_and_dispose(): TRACE: Socket 0 receive bytes 16 of 16
0026899671 [hal] socket_hal.cpp:792, read_packet_and_dispose(): TRACE: Socket 0 receive bytes 2 of 18
0026899671 [hal] socket_hal.cpp:792, read_packet_and_dispose(): TRACE: Socket 0 receive bytes 16 of 16
````
as well as our normal application-level logs

B. Lost WiFi connection, with our application sensing that `Particle.connected()` was no longer true. System outputs this (only time this ever happened in the log)

````
0026916843 [system] system_cloud_internal.cpp:1322, system_cloud_active(): WARN: Disconnecting cloud due to inactivity! 26916843, 26901831
0026916844 [system] system_task.cpp:427, cloud_disconnect(): INFO: Cloud: disconnecting
0026916844 [system] system_task.cpp:440, cloud_disconnect(): INFO: Cloud: disconnected
````

C. Our program is still running, but all debug logs stop, indicating the system threads are no longer correctly running. Our program attempts to fix the connection by trying a WiFi reset (turning the chip on and off, and resetting `Particle.connect()` and `WiFi.connect()` ) but to no avail. None of our calls bring us out of breathing green and nothing causes the debug logs to appear again, even as we make these calls to the system FW.

D. Our program stops running 21 minutes after (C). The serial session is still active, according to CoolTerm, but there is no new data.

Could it be that the crash at D is due to unanswered calls from our application to the System FW?

Either way, i hope this is useful data. I can privately send you the full log files as well, if you'd like to look for patterns.

pomplesiegel, 2016-06-14:
So for the debug logs of (B) to occur, 
````cpp
system_tick_t now = millis();
if (SPARK_CLOUD_CONNECTED && ((now-lastCloudEvent))>SYSTEM_CLOUD_TIMEOUT)
````
would have to have been true. 

But according to our logs, we had a successful function call just seconds before that occurred, so i'm confused why the system fw chose to disconnect.

monkbroc, 2016-06-14:
From this log we can guess that Spark_Process_Events which updates lastCloudEvent is no longer called. Backing out goes to handle_cloud_connection, manage_cloud_connection, Spark_Idle_Events and finally system_thread_idle which is the idle function of SystemThread. So SystemThread is stuck doing something else, but we have no indication what. Let me think what we could do as a next step.

monkbroc, 2016-06-15:
@pomplesiegel Here's what we can do next: let's try dumping the stack of the system thread when the error condition occurs to see where the system thread is stuck.

I wrote some utility code to walk through the FreeRTOS threads to pick the system thread (you don't create any `std::thread` in your application code, right?) and dump the stack to serial.

The idea is to wait for the error condition to occur (something like `Particle.connected() == false`) and dump the system thread stack to see what it is up to. On ARM we can't do an onboard stack trace of only function calls, but that info will be easy to glean from the full stack dump.

https://gist.github.com/monkbroc/6ef1bf003587a4c1bb773bd48a8efd9d

Please let me know if the instructions are not clear.

Good luck :muscle:

pomplesiegel, 2016-06-15:
@monkbroc, that's great! 

We do create a swTimer using the high-level particle interface, like this

````cpp
Timer t(1000, &CallbackClass::onTimeout, callback);
````

but do not explicitly create a `std::thread`

If that should still be fine, we'll get going on this today.

monkbroc, 2016-06-15:
That timer will not interfere.

DaliborFarny, 2016-06-19:
Got the programmer shield, now trying to get the backtrace output. I am on Win10, following instructions from @monkbroc (https://github.com/spark/firmware/issues/904#issuecomment-223349018 and https://medium.com/@jvanier/5-steps-to-setup-and-use-a-debugger-with-the-particle-photon-ad0e0fb43a34#.olae64wbf)

I remapped D3-D7 to other pins to be able to use JTAG properly. I am now playing with Photon inserted into programmer shield, running (not stuck at breathing green).

I am having problems with `bt` command (returns #0  0x00000000 in ?? ()), do I need to put breakpoints into the code to make it work?

> 

Output from openocd:

> C:\Particle\openocd\bin-x64>openocd -f particle-ftdi.cfg -f stm32f2x.cfg -c "gdb_port 3333"
Open On-Chip Debugger 0.9.0 (2015-05-19-12:09)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
jtag_ntrst_delay: 100
none separate
cortex_m reset_config sysresetreq
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : clock speed 1000 kHz
Info : JTAG tap: stm32f2x.cpu tap/device found: 0x4ba00477 (mfg: 0x23b, part: 0xba00, ver: 0x4)
Info : JTAG tap: stm32f2x.bs tap/device found: 0x06411041 (mfg: 0x020, part: 0x6411, ver: 0x0)
Info : stm32f2x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0x20036411
Info : flash size = 1024kbytes
undefined debug reason 7 - target needs reset

Output from DBG:

> C:\Particle\firmware>arm-none-eabi-gdb -ex "target remote localhost:3333" build/target/main/platform-6/my_app.elf
GNU gdb (GNU Tools for ARM Embedded Processors) 7.8.0.20150604-cvs
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/target/main/platform-6/my_app.elf...done.
Remote debugging using localhost:3333
0x00000000 in ?? ()
(gdb) info thread
[New Thread 536913536]
[New Thread 536930736]
[New Thread 536914136]
[New Thread 536920952]
[New Thread 536932200]
[New Thread 536923064]
[New Thread 536922248]
  Id   Target Id         Frame
  8    Thread 536922248 (worker thread) 0x0802604c in vPortYield () at WICED/RTOS/FreeRTOS/ver7.5.2/Source/portable/GCC/ARM_CM3/port.c:332
  7    Thread 536923064 (worker thread) 0x0802604c in vPortYield () at WICED/RTOS/FreeRTOS/ver7.5.2/Source/portable/GCC/ARM_CM3/port.c:332
  6    Thread 536932200 (WWD) 0x0802604c in vPortYield () at WICED/RTOS/FreeRTOS/ver7.5.2/Source/portable/GCC/ARM_CM3/port.c:332
  5    Thread 536920952 (Tmr Svc) 0x0802604c in vPortYield () at WICED/RTOS/FreeRTOS/ver7.5.2/Source/portable/GCC/ARM_CM3/port.c:332
  4    Thread 536914136 (app_thread) 0x0802604c in vPortYield () at WICED/RTOS/FreeRTOS/ver7.5.2/Source/portable/GCC/ARM_CM3/port.c:332
  3    Thread 536930736 (tcpip_thread) 0x0802604c in vPortYield () at WICED/RTOS/FreeRTOS/ver7.5.2/Source/portable/GCC/ARM_CM3/port.c:332
  2    Thread 536913536 (system monitor) 0x0802604c in vPortYield () at WICED/RTOS/FreeRTOS/ver7.5.2/Source/portable/GCC/ARM_CM3/port.c:332
* 1    Thread 536920456 (IDLE :  : Running) 0x00000000 in ?? ()
(gdb) bt
#0  0x00000000 in ?? ()
#1  0x00000000 in ?? ()
(gdb)

monkbroc, 2016-06-20:
It looks fine @DaliborFarny. Try `break setup`, `monitor reset halt`, `continue` and check that that it stops on `setup()`, then try `bt`.

monkbroc, 2016-06-20:
@pomplesiegel gave a big push to reproduce this issue and has found a short program to reproduce it by alternating calls to `Particle.publish` and `Particle.syncTime`! Thanks :100: :tada: :confetti_ball: 

I fixed an issue in the way the call to `Particle.syncTime` is dispatched.

@DaliborFarny I assume since your product is a clock you call that method too? Please try the `feature/sync-time-multithread` branch to see if the problem is fixed for you too.

https://github.com/spark/firmware/tree/feature/sync-time-multithread

DaliborFarny, 2016-06-21:
@monkbroc Great news! I built the firmware with the sync-time-multithread patch for one clock, it is running right now, I will observe it. Another clock is running 0.5.1 with JTAG connected, waiting for the issue to appear.

[9 more replies]

closed 2016-07-07 by technobly