particle-iot/device-os #915: applying our logging framework

m-mcgowan, 2016-03-18:
Now that 0.5.x includes a shiny new logging framework, we need to apply that framework to our product.

Here's a bucket of thoughts about how we can apply the logging framework:

- have a master switch, plus a on/off switch for each logging output so that logs can be easily disabled. (As an implementation detail, Ideally if all outputs are off then logging is a no-op, avoiding any runtime overhead.) 
- default logging outputs warnings and errors to USBSerial1
- a way to change the logging filter without writing code? e.g. if a user wants to enable debug logging for customer support
- tidy up logs - make them intelligible, self-explanatory
- documentation for logs, what they mean.
- logging API errors (we are currently quite weak here.) E.g. logging when a publish fails, or function registration fails because the variable name is made dynamically and is too long. 
- Spark.log()? https://github.com/spark/firmware/issues/252
- logging output to TCP/UDP
- logging output as published events? (Need to be careful here that this is only for events occurring seldomly, or have the log self limit in some way.)
- logging configuration over serial in listening mode/from CLI?
- build photon firmware with with DEBUG_BUILD=y so we don't have two versions of system firmware.

cc: @sergeuz @technobly @jenesaisdiq @monkbroc

sergeuz, 2016-04-22:
Guys, current documentation for logging can be found here: https://github.com/spark/docs/blob/feature/logging/src/content/reference/firmware.md#logging – I think it gives nice overview of the API. Below are some of the changes I wanted to cover along with the questions raised by Mat:

- While logging framework is backward-compatible with current API, consider using the newer API instead, as it makes the intent more clear: `LOG(INFO, …)` instead of `INFO()`, `LOG_FORMAT()` instead of `DEBUG_D()`, etc. Also, the internet is full of cries already due to WinAPI defining `ERROR` macro :)

- Logging is enabled by default for release builds. This increases size of the non-modular build by ~6k and doesn’t seem to be a problem, however, considering our goal to make logs more readable and user-friendly, I believe some of the current debugging output needs to be guarded by `DEBUG_BUILD` or `LOG_DEBUG_*()` macros.

- `DEBUG` and `LOG` severity levels were deprecated, as their precedence wasn’t clear without looking into the source code or docs every time. If `TRACE` level is not enough for verbose output then I suggest to use either separate category (e.g. `hal.modem.at` for AT commands log) or `LOG_DEBUG_*()` macros additionally.

- Category filtering adds some overhead if enabled at runtime, so it’s usually not a good idea to dump any data in byte-by-byte manner. Consider using `log_dump()` function instead.

zsup, 2016-04-22:
Hey guys - I apologize if this feedback is late to the game, but are you guys open to feedback to make the API more beginner-friendly? Specifically I'd like to recommend the following:

- Replacing capitalized macros with camelCase methods. For instance, replacing `LOG()` with `logger.log()`
- Adding a `log()` call that doesn't require setting the log level, and defaults to `INFO`. In other words, `logger.log("Hello, world!");` would be equivalent to `logger.log(INFO, "Hello, world!");`
- I find the logging categories confusing, and it might be a vocabulary issue. Can we call them topics instead? We talk about and use topics frequently with our pub/sub system, so I feel like that would be more familiar, although then I would also expect the syntax to be more like `Particle.publish()` syntax.
- Refactoring the documentation so that the aspects that will be used frequently by lots of people with varying skill levels are documented first (namely creating a logger, `log()`ing, setting the logging level), and the rest of the documentation for topics that are more advanced (categorizing, printing, dumping, etc.) are pushed later so that if beginners are confused by them they still get the basics
- Docs should also include an explanation of what to expect as output on your computer. If you log something, what do you have to do on the computer side to see that log, and how is it displayed?

Hope that's helpful feedback!

sergeuz, 2016-04-22:
Zach, thanks for the feedback.

> Replacing capitalized macros with camelCase methods. For instance, replacing LOG() with logger.log()

Macros are primarily used to perform automatic category substitution, e.g.:
```
LOG_SOURCE_CATEGORY("global") // Overrides LOG_MODULE_CATEGORY

void foo() {
    LOG_CATEGORY("foo");
    LOG(INFO, "Hello"); // Uses "foo" category
}

void bar() {
    LOG(INFO, "Hello"); // Uses "global" category
}
```
In principle, we can add simple wrapper over existent logging functions to the wiring library and make above code look like following in the user code:
```
Logger logger("global");

void foo() {
    Logger logger("foo");
    logger.info("Hello"); // Uses "foo" category
}

void bar() {
    logger.info("Hello"); // Uses "global" category
}
```

This will require renaming of the current `Logger`, `SerialLogger` and other classes into something like `LogAppender`, `SerialLogAppender` respectively.

> I find the logging categories confusing, and it might be a vocabulary issue. Can we call them topics instead?

The term "category" seems to be quite usual in the context of logging, for example: http://doc.qt.io/qt-5/qloggingcategory.html, https://logging.apache.org/log4j/1.2/apidocs/org/apache/log4j/Category.html. However, personally I don't have any strong preferences here.

andyw-lala, 2016-04-22:
I'll repeat my enthusiasm for a syslog plug-in.

http://community.particle.io/t/anyone-ever-use-text-of-udp-for-debug/13793

Note also RFC 3164 recommendation in that thread.

sergeuz, 2016-04-22:
@andyw-lala I cannot say for sure, but our next task here will probably be cloud-enabled logging. Same time, significant part of the logging now lives in user space, which makes implementation of the custom logger backends, such as syslog, more straightforward.

m-mcgowan, 2016-04-22:
I have syslog code from another project so we could make a library to provide syslog output.

m-mcgowan, 2016-04-22:
Just for visibility (we discussed briefly internally) what we have so far is the logging framework that is primarily intended to help organise our system firmware logging. We will add a nice API for application/library logging.

andyw-lala, 2016-04-22:
Cloud logging is great, and should obviously be the top priority, seamlessly for user and system events.

I'd love to see a syslog plug-in as an option (additional to , replacement for ?) out of the chute, as an immediately useful proof-of-concept & example for plug-ins.

You probably all know my mantra of "as soon as you have to hook up a wire, we lose our hardware distinction from arduino/teensy/etc", and are probably sick of it, but I still hold it to be true. Any time we can remove another use case for "hook up the USB port to a computer...", I'm all over that like a cheap suit. Those that haven't heard me say it, you can roll your eyes pre-emptively :)

I believe I see similar moves afoot for debugging, with a software gdb client, and I think that idea rocks too.

sidwarkd, 2016-09-11:
@sergeuz @zsup I know this thread is a little dated but think my comments apply here. I've been playing around with the new logging framework all evening and found something that seems a little funny that perhaps you can explain. I've found that LOG() calls honor the LOG_INCLUDE_SOURCE_INFO define but calls to Log.<type> do not. Digging through the code the reason for this is how the attributes are initialized through the different code paths. So currently the only way to get things like the file and function info in the log output is to use LOG() instead of Log(). Was that intentional and if so what is the reasoning behind it? Just curious. Thanks and keep up the great work.

sergeuz, 2016-09-12:
@sidwarkd It's by design. `LOG()` is a function-alike macro which expands at compile time and thus can use source info provided by the compiler at every invocation, while `Log()`, `Log.info()` and etc. are regular method calls compiled separately as part of the API library.

sidwarkd, 2016-09-12:
Thanks for the explanation @sergeuz. Love the logging stuff. Nice work.

closed 2020-02-29 by avtolstoy