particle-iot/device-os #925: [Photon] millis() jumps suddenly

pomplesiegel, 2016-03-26:
Yesterday we noticed an issue with our lighting fades (based on usage of `millis()`) wherein short fades were jumping suddenly to the end. After a deeper dive, it became clear that the fade was working properly, but the value returned by millis() would suddenly jump.

At first, I figured this was caused by #916, but that issue would cause millis values returned to simply not increment for a long period of time, correct? This looks as if there's suddenly a block when millis() is called, as without warning, the value will jump, without showing the same value over and over, as it would if that value were simply not incremented for a while. From our experience this jump can be up to 300ms, which will make for a very jarring transition. 

This can be seen by a simple application which prints the current `millis()` value:

#### Program to reproduce

Compiled against the tip of develop (as of today) for Photon, this easily reproduces the issue

````cpp
#include "application.h"

SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

void setup()
{
  Serial.begin(9600); //Begin USB serial session
  Serial.blockOnOverrun(false); //don't let serial block

  delay(2000); //start-up delay
  WiFi.off(); // no wifi distractions
}

void loop()
{
  Serial.println("Millis: " + String( millis() ) ); 
}
````

#### Sample output

Jump example 1 (small-ish):
````
Millis: 10430
Millis: 10430
Millis: 10430
Millis:Millis: 10431
Millis: 10432
Millis: 10432
Millis: 10432
Millis: 10432
Millis: 10432
Millis: 10432
Millis: 10432
Millis: 10432
Millis: 10432
Millis: 10432
MMillis: 10478 <<<<  46 ms jump
Millis: 10478
Millis: 10478
Millis: 10478
````

Jump example 2 (very large): 
````
Millis: 10502
Millis: 10502
Millis: 10502
Millis: 10502
Millis:Millis: 10503
Millis: 10503
Millis: 10504
Millis: 10504
Millis: 10504
Millis: 10504
Millis: 100824 <<<< that's a 320ms jump !!
Millis: 10824
Millis: 10824
Millis: 10824
````
....

#### Details

- Issue occurs with the WiFi chip ON or OFF
- Issue occurs with `SYSTEM_THREAD(ENABLED)` or `DISABLED`
-  Issue occurs with `Serial.blockOnOverrun(false)` or `true`

#### Issue

This is a very large issue for us, as much of our code relies on millis's accuracy. There are smaller sections which could be re-implemented with `System.ticks()`, but its frequent overflow is not ideal for all applications. 

What are your thoughts on this?

pomplesiegel, 2016-03-28:
#### Update
This appears to be more of a generic issue: the user application being blocked for hundreds of milliseconds at a time.

Using a completely different source for "millis", the same phenomenon occurs for both our actual application and this test app. 

Here we loosely implement `millis()` based on System.ticks()

#### Program to reproduce using alternate millis() implementation

````cpp
#include "application.h"

SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

unsigned long computedMillis()
{
   return System.ticks() / ( System.ticksPerMicrosecond() * 1000 );
}

void setup()
{
  Serial.begin(9600); //Begin USB serial session
  Serial.blockOnOverrun(false); //don't let serial block

  delay(2000); //start-up delay
  WiFi.off(); // no wifi distractions
}

void loop()
{
  Serial.println("Millis: " + String( millis() ) ); 
}
````

#### Sample output
````
Millis: 80348
Millis:Millis: 80349
Millis: 80349
Millis: 80349
Millis: 80349
Millis: 80349
Mi9
Millis: 80489 <<<< 140ms jump
Millis: 80489
Millis: 80489
Millis: 80489
Millis: 80489
Millis: 80490
````

#### In application
It appears as if the user application is simply locked from executing for hundreds of milliseconds at a time, which prevents us from performing steady fades, as we must now catch up for lost time.

technobly, 2016-03-29:
Just acknowledging your issue @pomplesiegel and wanted to share a few thoughts.

System.ticks() returns the DWT->CYCCNT value which is a 32-bit hardware counter, so good to know it does just appear that the main thread is blocked for periods of time.  249ms and 320ms seems like a long time in the scheme of things on the Photon, but a shorter time on the Electron.  It would be good to understand what is causing this time delay.  It might be worth looking into adjusting the priority level of interrupts to solve this for your fading purposes.  Perhaps a software based timer with a high priority will solve the issue.  First of all, try software timers as is: https://docs.particle.io/reference/firmware/photon/#software-timers  Then perhaps try them with an elevated priority level...  Perhaps @avtolstoy can comment on if we have discussed adding this type of a feature to Software Timers yet, similar to we did here for attachInterrupt() https://github.com/spark/firmware/pull/811/files

* [ ] understand the cause of main thread being blocked for 250ms or more
* [ ] test raising the priority level of Software Timers act as solution to give user firmware more real time control, is this a solution?
* [ ] add user selectable interrupt priority to Software Timers on the road map

pomplesiegel, 2016-03-30:
Hi @technobly, thanks for looking into this and providing some helpful tips. We'll be sure to give those a try and report back.

m-mcgowan, 2016-04-01:
Can you reproduce the issue with no external hardware attached, or perhaps identify which hardware is causing the delay? There have been reports of this previously that I was unable to reproduce, so great to have this one under the spotlight again!

closed 2016-08-09 by technobly