particle-iot/device-os #929: Error while compiling the Electron application Power

krvarma, 2016-03-31:
I am trying to compile the Electron application Power (firmware/user/applications/power/). I ma using the following command. I am in `main` folder

_make clean all PLATFORM=electron APP=power PARTICLE_DEVELOP=1_

But it is getting error

```
<firmware folder>\user/applications/power/application.cpp:51: undefined reference to `MDMParser::powerOn(char const*)'
<firmware folder>\user/applications/power/application.cpp:112: undefined reference to `electronMDM'
```

This error is getting for all applications using `extern MDMElectronSerial electronMDM;`

technobly, 2016-07-03:
The Power application is really just an old app that we used to test things with the PMIC and Fuel Gauge.  It's in a bad state right now, and it will compile easily with a few items removed... but it's unclear what it should really be doing anymore.  I.e., it's probably a bad idea to use it without a clear understanding of what everything is doing.  If there is something specific you are trying to do with the PMIC and Fuel Gauge, please open a community thread about it.  For now I'm going to remove the Power app to avoid future confusion.

closed 2016-07-03 by technobly

krvarma, 2016-07-04:
Thanks @technobly, I was trying to compile with a firmware which declares `extern MDMElectronSerial electronMDM;`. Any application with this declaration causes this issue. Simply put this declaration and the linking fails with undefined reference. I just given and example to start with.