particle-iot/device-os #936: Electron Software Timer Causes SOS

kenbiba, 2016-04-04:
The following program works on Photon but causes SOS on Electron when line 614 is not commented out starting the telemetryTimer.

// Start the timers for the telemetry, storage and LED flash events
/*
    telemetryTimer.start();                             // Start telemetry timer to send telemetry periodically
*/   
    flightReady.start();                                // Start LED with flight status based on 3D GPS lock
    saveTimer.start();                                  // Start storage save timer

/*****************************************************************
This is the core code of the 2016 S4 (Small Satelleites for Seconddary
Students) Particle (either Photon or Electron) based payload.

Ken Biba

9DOF sensor I2C
Baro Sensor I2C
Serial 0 (USB) - diagnostic
Serial 1 - GPS
Serial 2 - Storage

On Photon telemetry is via WiFi, on Electron cellular

Optional for LoRa radio

Save to storage happens at 10 Hz
Telemetry is currently 1 Hz based on Particle Cloud limitations of 1/event/second

*****************************************************************/
#include "SparkFunLSM9DS1/SparkFunLSM9DS1.h"
#include "math.h"
#include "SparkFun_MPL3115A2/SparkFun_MPL3115A2.h"
#include "Serial4/Serial4.h"
#include "SparkFunMAX17043/SparkFunMAX17043.h" // Include the SparkFun MAX17043 library
#include "TinyGPS++/TinyGPS++.h"

// Mode
#define DIAG        TRUE            // Print to serial port diagnostic stream
#define ELECTRON    

// Communication Configuration

#define CLOUD       TRUE            // Tranmit to Particle cloud 
#define WIFI        FALSE           // WiFi enable?
#define CELL        TRUE            // Cellular enable?
#define LORA        FALSE           // LoRa enable?   LoRa can be enabled in parallel with either WiFi or Cellular
#define SAVE        FALSE           // Local storage enable?

// Sensors configuration

#define IMU         TRUE            // Photon IMU sensors enabled
#define BARO        TRUE            // Baro sensor enabled
#define GPS         TRUE            // GPS sensor enabled
#define POWER       TRUE            // Battery sensors enabled
#define STORAGE     FALSE           // Local storage enabled

#define BAROMODE    TRUE            // Barometer mode.  False is altimeter mode
#define BAROSAMPLE  3               // Barometer oversampling rate.  Major overhead if much higher

#define STORAGESPEED    38400       // Speed of serial interface to storage for microSD card
#define STORAGETIME     100         // Time between state storage
#define GPSSPEED        9600        // Serial speed to GPS
#define TELEMETRYTIME   1000        // Milliseconds between telemetry messages.   For Electron at 1/second it is 1000.
#define FLIGHTTIME       500        // Milliseconds between flashes.   Blue flash indicates 3D GPS lock and ready for flight.

#define FLASHTIME       1000        // LED flash interval in milliseconds
#define LED     7

#ifdef PHOTON
#define SSID    "S4"                // SSID of WiFi access point when used with Photon
[... 797 lines cut ...]
    Serial.print(gps.altitude.meters(), 1);
    Serial.print(F("  Date/Time: "));
    Serial.print(gps.date.month());
    Serial.print(F("/"));
    Serial.print(gps.date.day());
    Serial.print(F("/"));
    Serial.print(gps.date.year());

    Serial.print(F(" "));
    Serial.print(gps.time.hour());
    Serial.print(F(":"));
    Serial.print(gps.time.minute());
    Serial.print(F(":"));
    Serial.print(gps.time.second());
    Serial.print(F("."));
    Serial.print(gps.time.centisecond());
    
    Serial.print(F(" "));
    Serial.print(gps.charsProcessed()); Serial.print(F(","));
    Serial.print(gps.sentencesWithFix()); Serial.print(F(","));
    Serial.print(gps.failedChecksum()); Serial.print(F(","));
    Serial.print(gps.passedChecksum());

    Serial.println();
}

m-mcgowan, 2016-04-04:
Allow me to suggest that you post this to http://community.particle.io - there's a lot of code here (almost 900 lines) to inspect for the cause of the issue.  Please only use github issues for focused known defects in the firmware or required features. 

You mention that the bug only occurs when the telemetry timer is started. The handler for the timer `sendTelemetry()` has code that is conditionally compiled for the electron. That code includes `Particle.publish()` call. This should not be executed from an interrupt context. (Please see the documentation on software timers for what is safe to call.)

As a starting point, you can first remove all the code from the `sendTelemetry()` function to determine if it's the timer itself that is failing, or the timer handler. If the fault doesn't occur when the timer is active, with an empty handler function, then we know it's not the timer itself, but a bug in the handler code.

I hope that gets you started. I'll close out this issue - please ask for additional help on the community forums. Thanks!

closed 2016-04-04 by m-mcgowan

m-mcgowan, 2016-04-05:
Hello again - I just discovered you already had a forum post and were directed to create an issue here. Apologies for bouncing you around between pillar and post, but rest assured that until we can narrow down the problem, the forum is the best place for discussion where we may be able to narrow down and pinpoint the issue more concisely before moving to a github issue here. 

https://community.particle.io/t/electron-sos-for-program-with-software-timers-works-on-photon-sos-on-electron/21726

kenbiba, 2016-04-05:
Thanks for updating.  

I think best advice you gave was not using particle.publish in the timer event.  I did not anticipate it was blocking but that makes sense even if unfortunate. 

I can do a workaround for Electron but still interesting that it works on Photon. 

K

Ken Biba
Novarum, Inc. 
415-577-5496