particle-iot/device-os #937: HTTP server does not handle a request across multiple packets

brycekahle, 2016-04-04:
Original issue #680

I've confirmed this is still an issue with at least the `/set` HTTP POST request for setting the claim code. Below is the request body. It works correctly when contained in a single packet (tested via echoing to `nc`), but when the body and headers are separate packets the claim code is not set in the DCT (confirmed via reading the memory over DFU), even though success is returned.

```
POST /set HTTP/1.1
Host: 192.168.0.1
Proxy-Connection: keep-alive
Content-Length: 80
Pragma: no-cache
Cache-Control: no-cache
Origin: null
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_10_5) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/49.0.2623.87 Safari/537.36
content-type: text/plain
Accept: */*
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.8

{"k":"cc","v":"111111111111111111111111111111111111111111111111111111111111111"}
```

### Wireshark capture of 2 packet request
Packet 1
```
POST /set HTTP/1.1
Host: 192.168.0.1
Connection: keep-alive
Content-Length: 80
Pragma: no-cache
Cache-Control: no-cache
Origin: null
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_10_5) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/49.0.2623.87 Safari/537.36
content-type: text/plain
Accept: */*
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.8

```
Packet 2
```
{"k":"cc","v":"111111111111111111111111111111111111111111111111111111111111111"}
```

brycekahle, 2016-04-04:
After modifying the original test python script, an important detail is that it works if the first packet is ACKed before the second packet is sent. This is how Python is working in the test script, but not the browser.

brycekahle, 2016-04-04:
I created a node test script to confirm this, It is available here https://gist.github.com/brycekahle/64507af54c041b9e043fbf983fe890f5

brycekahle, 2016-04-04:
This article confirms this behavior in most browsers: https://josephscott.org/archives/2009/08/xmlhttprequest-xhr-uses-multiple-packets-for-http-post/

brycekahle, 2016-04-04:
The response format is also quite unusual:
```
1
{
1
"
1
r
1
"
1
:
1
0
1
}
0
```

m-mcgowan, 2016-04-13:
The response is chunked transfer. Here's the full response:

```
HTTP/1.1 200 OK
Content-Type: application/json
Cache-Control: no-store, no-cache, must-revalidate, post-check=0, pre-check=0
Pragma: no-cache
Connection: Keep-Alive
Access-Control-Allow-Origin: *
Transfer-Encoding: chunked

1
{
1
"
1
r
1
"
1
:
1
0
1
}
```

m-mcgowan, 2016-04-13:
This turned out to be a memory use bug, keeping hold of a pointer after the memory was freed. I imagine the memory was being reused depending upon how the TCP packets are received.

The fix is safe, so I've scheduled for inclusion in the imminent release.

closed 2016-04-13 by m-mcgowan

jme783, 2016-04-13:
Cool Thanks @m-mcgowan ! Is the "imminent release" 0.5.0-rc2?

m-mcgowan, 2016-04-13:
yep!

krazineurons, 2016-10-30:
so does 0.51 has the fix resolved? The softap-js page is still citing this issue as a warning in using HTTP protocol.

PR #956 (2016-04-14): softap claim code fix
The problem and steps to reproduce it are described in issue #937.

fixes #937. The memory passed from the parse_json() call is only valid during that call, yet the SetValueCommand was caching these for use later. The fix is to copy the data rather than store the pointer.

---

Doneness:

- [x] Contributor has signed CLA
- [x] Problem and Solution clearly stated
- [x] Code peer reviewed
- N/A - API tests compiled
- [x] Run unit/integration/application tests on device
- N/A - Add documentation
- [ ] Add to CHANGELOG.md after merging (add links to docs and issues)
commit 3f85c9a1 (2016-04-14): fixes #937. The memory passed from the parse_json() call is only valid during that call, yet the SetValueCommand was caching these for use later. The fix is to copy the data rather than store the pointer.