particle-iot/device-os #938: tone on A7 available after deep sleep

m-mcgowan, 2016-04-05:
https://community.particle.io/t/does-pwm-stop-on-a7-after-stop-mode/21621

wesner0019, 2016-04-07:
Sleep mode also affects digitalWrite
https://community.particle.io/t/problem-with-wkp-a7-after-system-sleep/21658/2

wesner0019, 2016-04-07:
@m-mcgowan is there any quick fix for the 5.0 firmware release. we have product that relays on the tone to work after waking up. right now we have to reset the system which become messy

m-mcgowan, 2016-04-07:
What kinds of workaround have you tried?

wesner0019, 2016-04-07:
reinitialize the tone pin after it wakes up: pinMode(A7,OUTPUT);

but that didnt work. I wasnt sure what else to try

m-mcgowan, 2016-04-20:
We have a fix, but I've postponed that for 0.6 since I feel we need some time to consider if we need to maintain the existing behavior (where A7 becomes the wakeup pin) or if we should change the behavior, or provide flags. 

I believe this to be a long-standing bug - there since the initial release of the Photon. The change is here - https://github.com/spark/firmware/commit/17b25f6d69163a27b1360615dffa6688e32b0cc8. 

As a workaround you might try calling

```
 PWR_WakeUpPinCmd(DISABLE);
 pinMode(A7,OUTPUT);
```
after waking up.

closed 2016-06-16 by technobly

danpe, 2016-10-21:
I'm still having the issue while using the [AssetTracker WakeOnMove](https://build.particle.io/libs/57c486cdb900af2aed000640/tab/3_WakeOnMove.cpp) example.

`System.sleep(WKP, RISING, TIME_PUBLISH_BATTERY_SEC, SLEEP_NETWORK_STANDBY);`

When I used the 0.5.3 electreon firmware version it stopped waking up the device after an hour.
Now after updating to 0.6.0-RC2 it takes about 3-4 hours of no movement to stop waking up the device.

PR #948 (2016-06-16): WKP pin should not be enabled as a wakeup source unconditionally for STOP mode
Fixes #938 

Confirmed that after exiting STOP mode with some other pin specified as a wakeup source, WKP pin retains its configuration (e.g. tone() continues working).

If WKP is specified as wakeup source for System.sleep(), its pinMode will be changed to either INPUT, INPUT_PULLDOWN or INPUT_PULLUP depending on trigger condition. The pin will have to be reconfigured.

---

Doneness:

- [x] Contributor has signed CLA
- [x] Problem and Solution clearly stated
- [x] Code peer reviewed
- N/A - API tests compiled
- [x [... cut]