particle-iot/device-os #939: TLS sockets

m-mcgowan, 2016-04-05:
With TLS socket support, HTTPS would be relatively simple to implement in the application layer. 

Some goals:

- supported on Electron/Photon/P1
- uses same library (so we are not troubleshooting two different implementations)
- doesn't use closed-source implementations (rules out WICED.)
- leverages as much existing code as possible to reduce code size increase (e.g. tropicSSL AES and other crypto routines already included in the communications lib in system-part1.)

Secondary goals:

- TLS server support (flash space permitting?)

zsup, 2016-04-05:
how about using mbedTLS, which we're using for the DTLS implementation on the
Electron?
https://tls.mbed.org/
https://github.com/ARMmbed/mbedtls On Tue, Apr 5, 2016 at 4:39 PM Matthew McGowan <notifications@github.com>
wrote:
With TLS socket support, HTTPS would be relatively simple to implement in the
application layer.

Some goals:

 * supported on Electron/Photon/P1
 * uses same library (so we are not troubleshooting two different
   implementations)
 * doesn't use closed-source implementations (rules out WICED.)
 * leverages as much existing code as possible to reduce code size increase
   (e.g. tropicSSL AES and other crypto routines already included in the
   communications lib in system-part1.)

Secondary goals:

 * TLS server support (flash space permitting?)

m-mcgowan, 2016-04-06:
You're a mind-reader -  that's exactly what I had in mind!

zsup, 2016-04-06:
:+1: also maybe an opportunity to refactor the TCP protocol to use mbedTLS to improve code reuse

m-mcgowan, 2016-04-14:
#957

perotom, 2021-06-21:
I know this issue is quite old but I think quite important. Right now I am using the TCPClient quite often to download data to SPI flashes. Sadly I need to use HTTP as my application size is already quite big. Are there any plans to support TLS for the TCPClient class? It is a real pity that we cannot use mbed which is already bundled with device os.