particle-iot/device-os #947: Particle.variable("name", value) retrieving incorrect values

m-mcgowan, 2016-04-07:
```
int startstate = 0;
void setup ()
{
Particle.variable ( "startstate", startstate);
}
```

Retrieved value is gibberish. (134million).

Changing the call to 

```
Particle._variable ( "startstate", startstate);
```

Fixes the value.  So it looks like something quirky with the name length checking macro.

sergeuz, 2016-04-10:
Encountered the same issue today. I think `bool variable(const T &name, Types ... args)` accepts variable's object by value rather that a reference, causing pointer to temporary object registered in lookup table.

legoadk, 2016-04-10:
This also seems to cause a hard fault in certain circumstances. Changing Particle.variable to Particle._variable for the time being seems to be a good temporary workaround for the invalid data and associated hard fault.

closed 2016-04-14 by technobly