particle-iot/device-os #955: WiFi automatically connect to AP after WiFi.on()

XuGuohui, 2016-04-13:
Hi there,

In general, `WiFi.on()` would not connect photon to an AP until `WiFi.connect()` is called. If I make the photon out of the AP range, it will always try to connect to the AP. Then I turn on the WiFi using `WiFi.off()` and then turn on it again by `WiFi.on()`, photon launched to connect to the AP immediately , instead of staying at fading blue. I.e.:

case 1.WiFi.on() -> stay at fading blue
case 2.WiFi.on() -> WiFi.connect() -> connect failed and WiFi.off() -> WiFi.on() -> connect to AP immediately.

Is the case 2 normal or a bug?

I looked into the code, I think it is probably caused by not clearing the wlan watch dog(`CLR_WLAN_WD()`) when calling WiFi.off(), so that upon the WiFi.on() is called the watch dog fired in function `manage_network_connection()`. Am I right?

If this is not a potential bug, I will close this issue. Waiting for your respond. 

Kind regards

m-mcgowan, 2016-04-13:
If you don't want the system to connect to the AP, call WiFi.disconnect() and then WiFi.off().

XuGuohui, 2016-04-14:
I tried as you told(disconnect first and then turn WiFi off) but no luck. It still automatically tried connecting to AP all the same if I turn WiFi on again.

technobly, 2017-02-13:
Hi @XuGuohui, in reviewing this it's unclear what mode was being used and if Particle.connect() was called.  Based on my own testing, I can say I believe Particle.connect() was being called, and should have been disconnected with Particle.disconnect() before turning WiFi off/on again.  Please use this test app to confirm.

Please see notes based on mode used:

```c
// Given: if WiFi is turned OFF and then back ON...

// will not try to automatically re-connect if Particle.connect() was never called, or else requires Particle.disconnect() to be called in case #5
SYSTEM_MODE(MANUAL);

// will not try to automatically re-connect if Particle.connect() was never called, or else requires Particle.disconnect() to be called in case #5
SYSTEM_MODE(SEMI_AUTOMATIC);

// will try to automatically re-connect on case #5 if Particle.disconnect() is not called, WiFi.disconnect() not required.
SYSTEM_MODE(AUTOMATIC);
```

```c
#include "Particle.h"

#if SYSTEM_VERSION != SYSTEM_VERSION_v061RC2
#error "please test with 0.6.1-rc.2"
#endif

// Given: if WiFi is turned OFF and then back ON...

// will not try to automatically re-connect if Particle.connect() was never called, or else requires Particle.disconnect() to be called in case #5
SYSTEM_MODE(MANUAL);

// will not try to automatically re-connect if Particle.connect() was never called, or else requires Particle.disconnect() to be called in case #5
// SYSTEM_MODE(SEMI_AUTOMATIC);

// will try to automatically re-connect on case #5 if Particle.disconnect() is not called, WiFi.disconnect() not required.
// SYSTEM_MODE(AUTOMATIC);

#define CHECK_STATE(data, state) \
        if ((data) == (state)) { \
            Log.info("Event: %s", #state); \
            return; \
        }

void connectionStateChanged(system_event_t event, int data) {
    if (event == network_status) {
        CHECK_STATE(data, network_status_powering_off);
        CHECK_STATE(data, network_status_off);
        CHECK_STATE(data, network_status_powering_on);
        CHECK_STATE(data, network_status_on);
        CHECK_STATE(data, network_status_connecting);
        CHECK_STATE(data, network_status_connected);
        CHECK_STATE(data, network_status_disconnecting);
        CHECK_STATE(data, network_status_disconnected);
    } else if (event == cloud_status) {
        CHECK_STATE(data, cloud_status_disconnected);
        CHECK_STATE(data, cloud_status_connecting);
        CHECK_STATE(data, cloud_status_connected);
        CHECK_STATE(data, cloud_status_disconnecting);
    }
}

SerialLogHandler logHandler(LOG_LEVEL_WARN, {
    { "app", LOG_LEVEL_ALL }
});

uint32_t t = 0;
int step = 0;

void setup() {
    waitFor(Serial.isConnected, 10000);
    System.on(network_status | cloud_status, connectionStateChanged);
    // if desired, make sure credentials are cleared and set to the wrong password to force a non-connection
    // although you can also try this WITH credentials and see the same behavior much quicker.
    // play with including some of the parts of case #2 and #4 that are commented out.
}

void loop() {
    if (step && millis() - t < 5000) {
        return;
    }
    switch (++step) {
    case 1: {
        Log.info("-> WiFi.on()");
        WiFi.on();
        break;
    }
    case 2: {
        Log.info("-> WiFi.connect()");
        WiFi.connect();
        // Particle.connect();
        break;
    }
    case 3: {
        // delay to observe behavior
        break;
    }
    case 4: {
        Log.info("-> WiFi.off()");
        // Particle.disconnect();
        // WiFi.disconnect();
        WiFi.off();
        break;
    }
    case 5: {
        Log.info("-> WiFi.on()");
        WiFi.on();
        break;
    }
    case 6: {
        // delay to observe behavior (desire is for WiFi to only be on, not automatically connecting)
        break;
    }
    default:
        break;
    }
    t = millis();
    
    if (System.mode() == MANUAL) Particle.process();
}
```

---

Please close this issue if this resolves the behavior you were seeing.

XuGuohui, 2017-03-01:
Hi @technobly , your code works as expected. But the code below would produce the issue:

```
SYSTEM_MODE(MANUAL);

void setup() {
  Serial.begin();
  delay(5000);
  
  WiFi.on();
  Serial.println("Connecting to WiFi");
  WiFi.connect();
  
  delay(10000);
}

void loop() {
  if (!WiFi.ready()) {
    Serial.println("Turn WiFi off");
    WiFi.off();
    
    delay(3000);
    
    Serial.println("Turn WiFi on");
    WiFi.on();
    
    delay(10000);
  }
}
```

Any comments are welcomed!

avtolstoy, 2020-02-29:
Not reproducible with recent releases.

closed 2020-02-29 by avtolstoy