particle-iot/device-os #961: Cellular.resolve() to provide hostname lookup

m-mcgowan, 2016-04-14:
inet_gethostbyname() exists for the Cellular HAL, but we ideally need a high-level API.

How about `Cellular.resolve()` to parallel `WiFi.resolve()`?

https://community.particle.io/t/inet-gethostbyname-compile-error/22065
---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

harrisonhjones, 2016-12-29:
Hey team, it looks like this feature would have helped at lease one community member in the [forums](https://community.particle.io/t/new-library-papertrail/28105/). @rickkas7 came up with [this solution](https://community.particle.io/t/new-library-papertrail/28105/7). Is there a better way or can we (I volunteer) package this up and add a `Cellular.resolve()` function?

avtolstoy, 2017-01-24:
Implemented in #1227, released in `0.6.1-rc.2`.

closed 2017-01-24 by avtolstoy