particle-iot/device-os #963: isascii not declared in this scope

mikey0000, 2016-04-15:
< Please summarize your issue here.  Check [x] as many boxes below as you can before submitting your issue.  Thanks -- you rock! >

On macOSX El Capitan with gcc-arm-none-eabi-5_3-2016q1 I am getting the error 
`../wiring/inc/spark_wiring_character.h:49:31: error: 'isascii' was not declared in this scope`
`/wiring/inc/spark_wiring_character.h:104:21: error: 'toascii' was not declared in this scope`
running make clean all
---

Completeness:
- [ x] Minimum test case added
- [ x] Device, system and user firmware versions stated
- [ ] Particle confirmed

kunbo, 2016-04-25:
I also get the issue, how I can fix it?

m-mcgowan, 2016-04-25:
You can pull in the pull request linked above which contains the fix.

PR #964 (2016-04-26): arm gcc 5.3.1 compatibility
adds GNU suport (for isascii and toascii) and avoids duplicate symbol definitions on linux. Provides compatibility with arm gcc 5.3.1 and fixes #963

---

Doneness:

- [x] Contributor has signed CLA
- [x] Problem and Solution clearly stated
- [ ] Code peer reviewed
- [x] API tests compiled
- [ ] Run unit/integration/application tests on device
- [ ] Add documentation
- [ ] Add to CHANGELOG.md after merging (add links to docs and issues)
…
commit 1b90aab0 (2016-04-26): adds GNU suport (for isascii and toascii) and avoids duplicate symbol definitions on linux. Provides compatibility with arm gcc 5.3.1 and fixes #963
commit 3f9233f2 (2016-05-20): adds GNU suport (for isascii and toascii) and avoids duplicate symbol definitions on linux. Provides compatibility with arm gcc 5.3.1 and fixes #963
closed 2016-04-26 by m-mcgowan