particle-iot/device-os #967: Suggest adding Time.uptime()

andyw-lala, 2016-04-19:
Most of the examples in the forums that claim to report/monitor uptime, are actually just reporting Time.now().

While you can monitor the event stream to pick up device online events, and there is movement on the restart syndrome front, having a simple unsigned integer to report in publish messages that is maintained by the clock/time subsystem and consistent with other time values would be valuable, IMO.

Putting the requirement out there for discussion first.

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

monkbroc, 2016-04-19:
How would that be different than `millis()`?

andyw-lala, 2016-04-19:
Resolution and rollover time ?

andyw-lala, 2016-04-19:
If millis() were uint64_t, I'd use that / 1000. But sadly, it is not.

technobly, 2017-02-07:
millis() wraps after about 50 days... that's a good long up time, but I can see the advantage of another uint64_t counter that's 50*50 days.  systick that updates millis() behind the scenes could easily manage this.

m-mcgowan, 2017-10-22:
`System.uptime()` added with https://github.com/spark/firmware/pull/1393

closed 2017-10-22 by m-mcgowan