particle-iot/device-os #971: SoftAP mode persisting when setup complete

m-mcgowan, 2016-04-20:
https://community.particle.io/t/listening-mode-persisting-past-softap-setup/19928/6

Test case:

```
SYSTEM_MODE(SEMI_AUTOMATIC);

void setup()
{
    WiFi.listen();
}
```

- start up the photon - connect to the softAP, e.g. `curl http://192.168.0.1/device-id`
- connect via serial and send 'x' to cause listening mode to exit
- connect to the softAP, e.g. `curl http://192.168.0.1/device-id` - the AP network is still available

Workaround:

```
void setup()
{
    WiFi.on();
    WiFi.listen();
}
```

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

pomplesiegel, 2016-04-21:
I believe @karlhenricksen has seen this as well - shown as the SoftAP WiFi network still broadcasting while the Photon is breathing cyan, and successfully connected to a network.

karlhenricksen, 2016-04-21:
Yes, I've seen this twice but haven't been able to find a reliable way to reproduce and only occurred when doing some irregular steps. Here are my notes from the 2nd time:
```
2nd occurrence after sending incorrect pw (returns to blinking blue):
- allow the 10min timeout (returns to blinking green) **this is a user timeout to exit listening mode
- tried to reset up wifi (even though was in blinking green), follow normal setup process
- eventually it successfully connected (breathing cyan) 
==> so seems it connected and new wifi creds actually went through.
==> now it's in an in-between state where it's connected AND broadcasting?```

m-mcgowan, 2016-04-21:
The key trigger condition is that WiFi is off when entering listening mode. This caused the softAP tear down code to not be executed.

pomplesiegel, 2016-04-21:
Hmm - that's interesting. In our code we never turn the WiFi chip off, yet we saw this occur twice.

m-mcgowan, 2016-04-21:
In semi automatic mode the wifi chip starts in an off state, so if the first thing that happens is WiFi.connect() with no credentials, then it will enter listening mode with WiFi being initially off.

pomplesiegel, 2016-04-21:
Interesting! OK, looking forward to this fix.

m-mcgowan, 2016-04-21:
The fix is in the PR linked above if you'd like to pull and integrate it. It will be released with 0.6.0.

PR #972 (2016-04-27): softap persists on exiting listening mode if wifi module was off.
when the wifi module was not active (and so made active in the listen method) the `on_stop_listning()` method was not called due to short-circuit evaluation when `started` was false.  Swapping the order fixes this.  Fixes #971.

---

Doneness:

- [x] Contributor has signed CLA
- [ ] Problem and Solution clearly stated
- [ ] Code peer reviewed
- [ ] API tests compiled
- [ ] Run unit/integration/application tests on device
- [ ] Add documentation
- [ ] Add to CHANGELOG.md after merging (add links to docs and issues)
commit 51cf6161 (2016-04-27): when the wifi module was not active (and so made active in the listen method) the on_stop_listning() method was not called due to short-circuit evaluation when `started` was false.  Swapping the order fixes this.  Fixes #971.
commit a3da1f2c (2016-10-26): when the wifi module was not active (and so made active in the listen method) the on_stop_listning() method was not called due to short-circuit evaluation when `started` was false.  Swapping the order fixes this.  Fixes #971.
closed 2016-04-27 by m-mcgowan