particle-iot/device-os #973: Particle.connect() should not allow loop() to run until Particle.connected() is true

technobly, 2016-04-20:
Particle.connect() should not allow loop() to run until Particle.connected() is true

See issue https://github.com/spark/docs/issues/367 in Docs, make sure to update behavior when resolved.

## Test App compiled with v0.5.0-rc.2 on Photon
```
SYSTEM_MODE(SEMI_AUTOMATIC);

void setup() {
    pinMode(D7, OUTPUT);
    digitalWrite(D7, LOW);
    Particle.connect();
}

void loop() {
    digitalWrite(D7, HIGH);
    // D7 LED should not light until Particle.connected() is true, but it does.
}
```

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [x] Particle confirmed

m-mcgowan, 2016-06-07:
I hear your case, but I don't feel this is a bug. That's the intended behavior of `SEMI_AUTOMATIC` mode, that the system always runs setup then loop, independently of the cloud connection state.

If it were `AUTOMATIC` mode then `loop` shouldn't run until connected, and that is indeed the case. 

A simple workaround is to add this in `setup()`

```
    Particle.connect();
    waitUntil(Particle.connected);
```

To so summarize - the behavior is by design that semi-automatic mode doesn't stop the loop from running when the device isn't connected. 

The situation would be the same with multithreaded code too - the setup()/loop() functions execute independently of the cloud state. In cases where the app wants to wait for the device to be in a given state, the `waitFor` function is always ready for duty! 

The `Particle.connect(`)/`disconnect()` calls are not themselves blocking - they just set flags. These flags do affect what happens next time the background loop is executed, and that is when any delay is experienced (in single-threaded mode) to connect to the cloud.

m-mcgowan, 2016-09-13:
On review, I think this is right - in SEMI_AUTOMATIC mode, loop/setup should not be run until the cloud is connected while `Particle.connect()` has been called.

PR #1145 (2016-11-28): Particle.connect() should be blocking in SEMI_AUTOMATIC mode
Closes #973

Makes `Particle.connect()` blocking in `SEMI_AUTOMATIC` mode (when threading is disabled).

Relevant tests:
- `CLOUD_01_Particle_Connect_Blocks_In_SemiAutomatic_Mode_With_Threading_Disabled`
- `CLOUD_02_Particle_Connect_Does_Not_Block_In_SemiAutomatic_Mode_With_Threading_Enabled`
- `CLOUD_03_Restore_System_Mode`

This PR also enables `wiring/no_fixture` to be compiled with threading enabled.

---

Doneness:
- [X] Contributor has signed CLA
- [X] Problem and Solution clearly stated
- [x] Code peer reviewed
- [x] API tests [... cut]
closed 2016-11-28 by technobly