particle-iot/device-os #978: Photon occasionally exits Listening mode prematurely (valid credentials are stored)

karlhenricksen, 2016-04-22:
#### Description:
With 0.5.0rc1 & 0.5.0rc2, I've seen 3 separate occasions where the Photon leaves listening mode (by holding setup button) after a minute or two, and once after only 15sec or so. This was never seen in the 0.4.9 release. In all cases (I think) we had credentials stored, most likely valid credentials (return to breathing cyan).

#### Steps to Reproduce:
**(not currently 100% reproducible)**
NOTE: we don't know the exact steps to reproduce since it's only occurred a few times, but it looks real now. Will have to update with more details as they come, but the steps in theory are below:

1. Ensure valid credentials are stored and Photon is connected to cloud (breathing cyan)
2. Hold setup button for 4s to put into listening mode
3. Wait for 1-2 minutes
4. Repeat steps 2-3

#### Version:
0.5.0rc1/rc2 system

#### Setup:
SYSTEM_MODE(MANUAL)
SYSTEM_THREAD(ENABLED)

karlhenricksen, 2016-04-22:
@m-mcgowan Have you seen this, or have any clues on how we might reproduce this? Maybe some way to get into a state where a premature timeout could occur?

technobly, 2016-04-23:
Hi @karlhenricksen there has been a feature in of the SETUP button for a while now that allows you to enter Listening Mode by holding SETUP for 3 seconds.  Then if you let go and press SETUP again and hold for >4s but less than 10s (10s erases credentials) and let go of SETUP, it will exit Listening Mode.  This allows you to exit Listening Mode without using the 'x' command from a serial terminal, or having to press RESET.  Is this possibly the behavior you are noticing?  It might be a bit intermittent if you are holding for exactly 4 seconds.  Try a 5 count.

karlhenricksen, 2016-04-23:
No, (though thats a nice feature), in this case it is just photon sitting in Listening mode. No interaction - usually I am going into wifi setup to send new credentials when I notice the photon has exited.

technobly, 2016-04-25:
If you keep performing steps 2 and 3, it will potentially make the device exit Listening Mode.  Are you seeing the device exit listening mode without any user interaction?  If so would you please revise your steps to reproduce above?  Thanks!

karlhenricksen, 2016-04-25:
Good catch, yes, updated to be more accurate to what I have seen. Unfortunately, it's not fully reproducible as of yet, and we've only seen it in passing - putting into listening mode and on 3 occasions the Photon exited without any interaction from user.

m-mcgowan, 2016-04-25:
Could you try in safe mode? This will rule out any interactions from the application and a potential trigger.

Here's an overview of the conditions that cause safe mode to exit:

- `WiFi.listen(false)` is called
- The setup button is pressed
- 'x' is sent over serial
- SoftAP connection sends the connect-ap command

Note also that with multithreading, the application loop continues executing during listening mode. You might wrap your main loop() logic in `if (!WiFi.listening()) { }` to rule out any interaction from the application while in listening mode.

sergeuz, 2016-04-25:
If you’re using linux then the problem could be caused by the ModemManager service which probes serial ports in order to detect modem devices: https://mail.gnome.org/archives/networkmanager-list/2013-October/msg00038.html

pomplesiegel, 2016-04-26:
Hi @karlhenricksen and @m-mcgowan, I believe our issue here may be due to the system_event's data not being correct. This is the only time we ever call `WiFi.listen(false)` in our application:

````cpp
//this will only be called by Particle if we're in listening mode
//This will check if we've been in listening mode for too long
//In this case, "data" returned by Particle will be for how many
//milliseconds we've been in listening mode
void HandlerForListeningModeCheckIn( system_event_t event, int data ) 
{
  //if we've been in listening mode for less than 10 minutes, just return
  if( data < 600000 )
    return;

  //otherwise, if we have credentials, try to exit listening mode, as this may
  //have been entered by mistake. Let's try connecting to one of our stored networks.
  if( WiFi.hasCredentials() )
  {
    Serial.println("!!!! Exiting listening mode after " + String(data) + " millis !!!!");
    WiFi.listen(false);
  }
}
````

within `setup()` we register this function using the call
````cpp
//register our callback function for checking up on listening mode
System.on(setup_update, HandlerForListeningModeCheckIn); 
````

Based on this logic, the only way we would call `WiFi.listen(false)` early is if `data` falsely reports a value larger than 600000 (milliseconds) too early. 

Does this make sense?

m-mcgowan, 2016-04-26:
Can you show the code for how you register that event listener?

m-mcgowan, 2016-04-26:
What hardware do you have attached? I suspect a source of interrupts is causing `millis()` to be incorrect, since you've previously reported in #925  that millis() is incrementing incorrectly for you.

pomplesiegel, 2016-04-28:
Our only application-instated interrupts are tied to 3 digital inputs on Photon - all filtered signals from a rotary encoder. These interrupt handlers are extremely brief.

We'll try to reproduce this w/ a completely separate application and report back.

kichooo, 2016-09-20:
After some time of inactivity I try to get my photon back to work again. After fresh restart, I hold setup to go into listening mode. It stays there for about 5 seconds, then starts blinking green. It will not let me turn listening mode again until next restart.

closed 2020-02-29 by avtolstoy