particle-iot/device-os #981: Particle.function returning float or double (C++ class member)

elevatorguy, 2016-04-25:
Not a big deal, as we can attach a global function returning a double to Particle.function

But in classes:

> ```
> // EXAMPLE USAGE WITH C++ OBJECT
> class CoffeeMaker {
>   public:
>     CoffeeMaker() {
>       Particle.function("brew", &CoffeeMaker::brew, this);
>     }
> 
>     double brew(String command) {
>       // do stuff
>       return 1;
>     }
> };
> 
> CoffeeMaker myCoffeeMaker;
> // nothing else needed in setup() or loop()
> ```

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

m-mcgowan, 2016-05-05:
To make this work requires more than just a syntactic change in firmware - the function registration in the cloud would need to be changed too so that the `double` is properly treated as a floating point number rather than as a 32-bit integer.

technobly, 2017-02-13:
This compiles, but currently returns a truncated result due to conversion to 32-bit integer as Mat mentioned.  I'm not sure why we allow this to compile though without error if it doesn't do what it looks like it should do.

```c
double brewGlobal(String command) {
    // do stuff
    return 3.14159;
}

void setup() {
    Particle.function("brew-global", brewGlobal);
}
```

closed 2020-02-29 by avtolstoy