particle-iot/device-os #985: WiFi.RSSI() stops working on Core in v0.4.9 & v0.5.0

GregSawin, 2016-04-27:
Calls to WiFi.RSSI() stop working shortly after the Core starts up and will return 2 (or 0 for others with this problem).  This doesn't appear to be an issue for Photons nor on Cores using v0.4.7 and below.  Thread on this issue with example code: https://community.particle.io/t/wifi-rssi-error-with-core-firmware-v-0-4-9/19819/7

Here's the last test code I used when I was trying to find the time threshold where it breaks down.  The 1st call always works, the 4th+ always returns 2 and calls 2 & 3 sometime work.  This is per startup, each subsequent loop returns all 2s for me.  You could greatly simplify this to 1 call/publish per loop and see the same results.
```
int rssi1 = 0;
int rssi2 = 0;
int rssi3 = 0;
int rssi4 = 0;
void setup() {
}

void loop() {
    rssi1 = WiFi.RSSI();
    delay(100);
    rssi2 = WiFi.RSSI();
    delay(200);
    rssi3 = WiFi.RSSI();
    delay(400);
    rssi4 = WiFi.RSSI();
    Spark.publish("test v0.4.9", "Wifi RSSI: " + String(rssi1) + " & " + String(rssi2) + " & " + String(rssi3) + " & " + String(rssi4), 60, PRIVATE);
    long msMillis = millis();
    while(millis() - msMillis < 8000)
        Particle.process();
}

```
---

Completeness:
- [x ] Minimum test case added
- [x ] Device, system and user firmware versions stated
- [ ] Particle confirmed

m-mcgowan, 2016-04-28:
Confirmed regression between 0.4.7 and 0.4.9.  

A temporary workaround is to disconnect WiFi then reconnect. 

The RSSI code hasn't changed between these two versions showing there regression, although I do see some behavior differences, such as the WiFi starting up on the Core when in SEMI_AUTOMATIC mode. (It should stay breathing white until WiFi is enabled in code.)

GregSawin, 2016-07-09:
Just confirmed this is still an issue on 0.5.1 and 0.5.2

closed 2020-02-29 by avtolstoy