particle-iot/device-os #986: Make clean all erases all build/target files!

pkourany, 2016-04-28:
The latest develop branch make files are erasing the build/target directory after compiling.  Running `make clean all PLATFORM=photon PARTICLE_DEVELOP=1 APP=myapp` from either the `modules` or `main` directories produces:

```arm-none-eabi-objcopy -O ihex ../../../build/target/user-part/platform-6-m/telnet.elf ../../../build/target/user-part/platform-6-m/telnet.hex
arm-none-eabi-size --format=berkeley ../../../build/target/user-part/platform-6-m/telnet.elf
   text    data     bss     dec     hex filename
  33612     224    2272   36108    8d0c ../../../build/target/user-part/platform-6-m/telnet.elf
arm-none-eabi-objdump -h -S ../../../build/target/user-part/platform-6-m/telnet.elf > ../../../build/target/user-part/platform-6-m/telnet.lst
make[1]: Leaving directory `C:/SPARK/ParticleBeta/firmware/modules/photon/user-part'
rm -f -r ../build/target
make: Nothing to be done for `all'.```

Running the same command but first with only `clean` then only with `all` produces the expected results (build/target directory intact).

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [ ] Particle confirmed

towynlin, 2016-05-03:
I saw this today, Particle has confirmed. 🙇

towynlin, 2016-05-03:
Verified that built products are now present after `make clean all` using commit cde57d1 in any of firmware root dir, main, or modules. 👍

m-mcgowan, 2016-05-05:
Thanks for verifying! closing out....

closed 2016-05-05 by m-mcgowan