particle-iot/device-os #988: po-util: A handy script for installing and using the Particle toolchain

nrobinson2000, 2016-04-30:
Hi guys,
**[I have been working on a script that allows users to easily install and use the Particle toolchain on OSX and Ubuntu based distros.](https://github.com/nrobinson2000/po-util)**  It takes care of installing dependencies and can be used to quickly compile and flash code to devices using dfu-util.  The script downloads and installs: [dfu-util](http://dfu-util.sourceforge.net/), [nodejs](https://nodejs.org/en/), [gcc-arm-embedded](https://launchpad.net/~terry.guo/+archive/ubuntu/gcc-arm-embedded), [particle-cli](https://github.com/spark/particle-cli), and the [Particle Firmware source code](https://github.com/spark/firmware).

This script works well, and I have tested this script many times on my Macbook and Linux Mint desktop, and I also have tested the script with Travis CI (Ubuntu and OSX) and Circle CI (also Ubuntu).

[image] [image]

I have been using this script for the past month for my own projects, and is very convenient.

Here is an overview of the what the script can do.
```
po-util Copyright (GPL) 2016  Nathan Robinson
This program comes with ABSOLUTELY NO WARRANTY.
Read more at http://bit.ly/po-util

Usage: po DEVICE_TYPE COMMAND DEVICE_NAME
       po DFU_COMMAND

Commands:
  build        Compile code in "firmware" subdirectory
  flash        Compile code and flash to device using dfu-util
  clean        Refresh all code
  init         Initialize a new po-util project
  patch        Apply system firmware patch to change baud rate
  update       Download latest firmware source code from Particle
  upgrade      Upgrade system firmware on device
  ota          Upload code Over The Air using particle-cli

DFU Commands:
  dfu         Quickly flash pre-compiled code
  dfu-open    Put device into DFU mode
  dfu-close   Get device out of DFU mode
```
I created this script because I needed an easy way to quickly compile firmware for my Particle projects.  I was able to install the Particle toolchain for myself, but I found it difficult to follow and decided that there needed to be an easier way for new users to install and use the Particle toolchain.

Check out my repository at: https://github.com/nrobinson2000/po-util

Check out the po-util webpage at: https://nrobinson2000.github.io/po-util/

Completeness:
- [x] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

nrobinson2000, 2016-04-30:
I think it would be good to have this script mentioned in the firmware documentation so that other users could benefit from it.

zsup, 2016-04-30:
This is great! @monkbroc is this complementary with the stuff you've been
working on re: an installer?

monkbroc, 2016-05-02:
This is very interesting. I wrote a self-contained version of the particle CLI a little while ago https://github.com/monkbroc/particle-cli-ng

This script is very comprehensive and installs a lot of tools. What we need to do is figure out what different users need at different points in the development cycle and help them install only what they need.

nrobinson2000, 2016-05-02:
The goal of my script is to provide an easy installation experience for new users.  I chose to make it install dfu-util, nodejs, gcc-arm-embedded, particle-cli, and the Particle Firmware source code so that users won't have to deal with installing the tools manually.

ScruffR, 2016-07-29:
I don't think this is wrong here, since it's not a firmware issue

technobly, 2017-02-07:
Working on a way to integrate into Docs here
https://github.com/spark/docs/pull/538
https://github.com/spark/docs/pull/417

closed 2017-02-07 by technobly