particle-iot/device-os #989: Error Compiling Firmware:  this program has been built without plugin support

mrmowgli, 2016-05-01:
Firmware won't compile as per the main README.md, after switching to 'latest' branch:

/particle/firmware/modules$ make clean all PLATFORM=photon -s program-dfu
  /home/mowgli/particle/firmware/modules/photon/system-part1/makefile /home/mowgli/particle/firmware/modules/photon/system-part2/makefile /home/mowgli/particle/firmware/modules/photon/user-part/makefile
sorry - this program has been built without plugin support
make[2]: *** [../build/target/communication/platform-6-m-prod-6/libcommunication.a] Error 1
make[1]: *** [communication] Error 2
make: *** [/home/mowgli/particle/firmware/modules/photon/system-part1/makefile] Error 2

nrobinson2000, 2016-05-01:
I see that you have starred and forked my script for installing and using the Particle toolchain. https://github.com/nrobinson2000/po-util 

Does it work for you?

mrmowgli, 2016-05-01:
I am abstracting your script a bit, partially because I have a lot of this
already installed  (Like Node and the Arm toolchain) but also your script
assumes a particular install path. I like it so far though!

nrobinson2000, 2016-05-01:
Thanks.  If you have any suggestions for improvement feel free to make a pull request.

nrobinson2000, 2016-05-02:
Hi @mrmowgli,
I have updated my script so that it lets the user choose the install path.

mrmowgli, 2016-05-02:
@nrobinson2000 Awesome!  Definitely helps.

Let me know if any of the other parts of the pull request are interesting.

nrobinson2000, 2016-05-02:
Thank you @mrmowgli.  I have updated my script with your changes and given you credit.

m-mcgowan, 2016-05-05:
Does that resolve the issue? I don't believe that error comes from our build system, and I'm unable to reproduce the issue so I'd need more details about your setup to investigate further.

mrmowgli, 2016-05-07:
Sorry haven't had a chance to check.  Will do it this weekend.

mrmowgli, 2016-05-23:
Ok, I switched to a different branch (I was using latest, but switched to a release candidate.) and I'm breaking earlier on what looks like a C++11 error, `can't find <functional>`.  I suspect it's because I have three versions of eabi-arm-none kicking around for different processors.  You can close this out.

m-mcgowan, 2016-05-25:
Closing as requested.

closed 2016-05-25 by m-mcgowan