particle-iot/device-os #990: [Core] FreeRTOS Software Timers cause hard fault

ScruffR, 2016-05-03:
In [this thread](https://community.particle.io/t/sos-status-after-using-the-software-timer-on-spark-core/22621) a member reported problems with Software Timers on Core 0.5.0.
After ticking off the usual suspects it turned out that not even the `timers.ino` sample provided with the `freertos4core` library works with 0.5.0 nor 0.4.9 anymore (0.4.7 works).

My own minimal test code looks like this
```cpp
#include "freertos4core/freertos4core.h"

void dmy()
{
    digitalWrite(D7, !digitalRead(D7));
}

Timer t1(1000, dmy);

void setup()
{
    pinMode(D7, OUTPUT);
}
```
and causes a hardfault despite not even calling `t1.start()`.

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [ ] Particle confirmed

wgbartley, 2016-06-03:
I just tried this out today.  The Core hard faults even if `freertos4core.h` is included (via particle CLI cloud compile) even without a timer being set up.

ScruffR, 2016-06-03:
Finally someone taking notice of this ;-)

monkbroc, 2016-06-04:
I love debugging those OS crashes. Dig deep into some multithreaded stuff!

nikojarvinen, 2016-06-18:
Not sure if this is related but I just posted simple software timer problem on electron: https://community.particle.io/t/software-timers-dont-work-on-electron/23809

monkbroc, 2016-06-20:
As @hossboss mentioned in https://github.com/m-mcgowan/particle-freertos4core/pull/4, the issue was caused by a stack overflow. The fix is to increase the stack size in freertos4core.

closed 2016-06-20 by monkbroc