particle-iot/device-os #992: I2C multi-master capability

XuGuohui, 2016-05-05:
Hi there,

Quote from STM32F205 reference manual:

> By default, it operates in slave mode. The interface automatically switches from slave to
> master, after it generates a START condition and from master to slave, if an arbitration loss
> or a Stop generation occurs, allowing multimaster capability.

Now I initialize the I2C module as slave by passing the own slave address when `Wire.begin(3)` and then invoke the master related methods to address and write data to a slave device. It hangs after the start signal issued, with solid white color of the RGB.

```
SYSTEM_MODE(MANUAL);

void setup() {
  // put your setup code here, to run once:
  Wire.begin(3);
}

void loop() {
  // put your main code here, to run repeatedly:
  Wire.beginTransmission(3);
  Wire.write(0x55);
  Wire.endTransmission(1);
}
```

I looked into the I2C HAL source code and found nothing special to cause this problem. Every code snip seams that it should work even though I initialize it as I2C slave to call the master methods. Can anyone pulls me out?

BTW, is it possible to implement the multi-master as documented in the reference manual? 

Thanks in advance!

---

Completeness:
- [x] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

avtolstoy, 2016-05-05:
Despite the fact that the microcontroller supports operation in both Master and Slave modes simultaneously (which is required for multi-master), I2C HAL implementation doesn't support that and would require significant overhaul. I'm not sure how high that is on the priority list.

The main issue here is that Master mode is implemented in a polling fashion, whereas Slave mode is interrupt-driven. When initialized in Slave mode, interrupts are enabled and status registers are checked inside the interrupt handler. In this case, `Wire.endTransmission(1);` doesn't see any state changes, because some states were already somehow handled (possibly incorrectly, due to Master/Slave mode mismatch) in the interrupt handler or some status flags were cleared just by reading the status registers inside the interrupt handler.

A possible workaround (untested) that could be used for your case:
```C++
SYSTEM_MODE(MANUAL);

void setup() {
  // put your setup code here, to run once:
  Wire.begin(3);
  pinMode(D2, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:

  // Either ATOMIC_BLOCK or NVIC_DisableIRQ/NVIC_EnableIRQ should work
  // NVIC_DisableIRQ(I2C1_EV_IRQn); // Disable I2C1 EV IRQ
  ATOMIC_BLOCK() {
    Wire.beginTransmission(5);
    Wire.write(0x55);
    Wire.endTransmission(1);
  }
  // NVIC_EnableIRQ(I2C1_EV_IRQn); // Re-enable I2C1 EV IRQ
  digitalWrite(D2, HIGH);
  delay(300);
  digitalWrite(D2, LOW);
  delay(300);
}
```

XuGuohui, 2016-05-05:
@avtolstoy Cool! your code works perfectly for me. Now I think I know why it hanged using my code. Thank you so much! Also looking forwards to the significant overhaul on I2C driver!

avtolstoy, 2016-05-05:
@XuGuohui, glad that it worked!

I might have phrased it a bit incorrectly :) There were quiet a few improvements to the current I2C HAL and it should be rather stable in both Slave and Master mode. I'm not sure that it needs the overhaul at the moment at all.

Should we keep this issue open to see if there is any interest in support for I2C multi-master mode? Or simultaneous master-slave / multi-master-slave modes?

XuGuohui, 2016-05-05:
OK, I'm glad to reopen it to let's vote for the improvements :)

closed 2020-02-29 by avtolstoy